UnchartedBull/OctoDash #25: Help

daimein2007, 2019-07-15:
pi@octopi:~/OctoprintDash $ npm install
-bash: npm: command not found

not sure what I'm doing wrong following instructions to a tee.

Building the App
Clone the repository - git clone https://github.com/TimonGaebelein/OctoprintDash && cd OctoprintDash
Install all the dependencies - npm install
Adjust src/assets/config.json accordingly to your installation
Build the app - ng build --prod

daimein2007, 2019-07-15:
Don't use Linux much but love this plugin /software just trying to get it installed I have the HTTP:// config done the install and the build does nothing sudo or no sudo, Not sure what im ither not doing or doing wrong. Using putty as admin in Octoprint.

UnchartedBull, 2019-07-19:
Hi, sorry I've just seen this issue. I wouldn't recommend going with the WebServer route, but rather use the electron app.

The issue in your case is, that you don't have node installed (which includes npm). So you need to install node first (either via https://github.com/nvm-sh/nvm or via the download from the node website). If you just want to run the app use the Electron App, if you already have a desktop installed it's a piece of cake ;)

UnchartedBull, 2019-07-19:
I'm going to close this now. If there are any issues with the Electron App let me know. It is definitely not recommended to use the webserver, as a later version might drop support for it.

closed 2019-07-19 by UnchartedBull