UnchartedBull/OctoDash #95: Octoprint Plugin

StefanCohen, 2019-08-19:
Octoprint Has the Temperature, Control, GCode Viewer and Terminal tabs but it doesn't really have a default landing page that provides an overview over the current state of the printer like OctoDash does.
It would be really cool of OctoDash could be bundled and distributed as an Octoprint plugin that registers as a new tab like this:

[image]

UnchartedBull, 2019-08-20:
That's a need idea. It requires some tweaking, but it should be possible. I'll focus on the release of OctoDash first, but after that is done it definitely is worth a look! Thanks for the suggestion!

StefanCohen, 2019-08-21:
Cool. I've been playing around with the layout a bit: 
[image]

Top-down, L>R:

(printer section, always displayed)
Printer Profile
Connection status
Printer Status

(Job section, only displayed if printing or paused)
Job progress
Time left (leveraging printtimegenious if installed)
Layer (or height, depending on user setting)
Estimated total print time (leveraging printtimegenious if installed)
Estimated material to be used in m (or g, depending on user setting)

(Sensor section, displayed depending of connection status)
Toolhead(s) Temp
Bed Temp
Fan Speed

UnchartedBull, 2019-08-22:
This looks really good. The design makes more sense for the Website, it requires however a almost complete rewrite of the UI. Therefore it may take quite some time for me to do it. I'll focus on the Raspberry Dashboard first and then have a look at extending the solution.

If you like - go ahead and start doing this deviation of OctoDash. I'll be happy to help you with that and will definitely will merge this into master!

StefanCohen, 2019-08-22:
Ok. I'll research what it would take to do as a native plugin. Beyond what is needed for the basic functionality I think it is necessary that it plays nice with other plugins such as themify and taborder etc.

StefanCohen, 2019-08-26:
Hi again!

I have started the development of a native plugin now. You can follow the progress here if you're interested: https://community.octoprint.org/t/dashboard-plugin/11536/18

UnchartedBull, 2019-08-26:
Looks great so far! You can get the FanSpeed and LayerHeight from the DisplayLayerProgress Plugin by OllisGit. He added an API for OctoDash, so you should be able to utilize that. If you need some help or styling tips just let me know :) If you like I can create a small mock-up with the look and feel of OctoDash, but it already looks pretty good!

StefanCohen, 2019-08-26:
Thanks. I'll take a look at DisplayLayerProgress. I have not been using the API so far but instead relied on the native viewModels but it is probably possible to create a dependency to another plugin like DisplayLayerProgress  and use its viewModels. I'll look into that.

I have very little experience with web development so I'm grateful for any assistance.

UnchartedBull, 2019-08-26:
This will be even better, as it will remove load from the Pi.

As I said the design looks good so far, I, personally, however would break up that strict tabular layout. It get's tiring to the eye if everything looks the same and some information may be hard to see at a glance and requires some getting used to. I've created a really really simple lookup of something I could image may look a bit more pleasing to the eye. You can use this as an inspiration to further develop the Dashboard :)

[image]

StefanCohen, 2019-08-26:
That's really good feedback and I like it. Colours are tricky as I have to account different Octoprint themes but I think there are ways to account for that. I'll look into the progress graph next as I think that's central to the dashboard. 

Btw, what type of display do you recommend for OctoDash? Are there many to choose from that has touch input?

StefanCohen, 2019-08-27:
I've added a progress bar for now:

[image]

UnchartedBull, 2019-08-27:
Looks great! If you can‘t get the theme colors to work, you can also think of putting these in the config. So everybody can adjust them to their liking.

The workflow looks good, but with JS you just can fire 4 asynchronous request at one time. Shouldn‘t be much overload and loading might be a bit quicker.

Yes there are many 😅 OctoDash runs best on a 800x480 screen, 480x320 screens are supported as well though. I would recommend a 5 inch screen, that connects via HDMI (so you get 60Hz) or the official RPi Screen. It depends on how much space you have though. I wouldn‘t go smaller than 3.5 inch as it gets very fidly. I think 5 inch is the sweetspot here, I‘m currently using a 7 inch and that’s a tad too big for me

UnchartedBull, 2019-08-27:
I think you can get a 5 inch one from China (AliExpress, Banggood, ...) for like 20 bucks. I may add some links to the wiki tomorrow :)

StefanCohen, 2019-08-27:
Cool. I'll order a screen after the wiki update 😃 

You can try the latest incarnation of the dasboard by installing the plugin from this link in the plugin manager: https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

UnchartedBull, 2019-08-27:
https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices Here are the links to the displays :). 

Cool I'll try that out, once I got some spare time. I think we can close this issue now, because it is being solved by you?

StefanCohen, 2019-08-27:
Thanks for the links. Yes. I'll close this now and will get back once it is somewhat stable in case you want to try it.

closed 2019-08-27 by StefanCohen

UnchartedBull, 2019-08-27:
That would be great! I will be more than happy to include the link to your project in the README once you‘re finished so everyone else sees this plugin as well :)