UnchartedBull/OctoDash #117: Integration Octoprint Plugin "Cancel Object" would be useful.

AndreasBrauckmann, 2019-09-02:
The integration of the plugin would be useful. Since plugin runs stable, in the GCode the labels have to be generated. These can be activated in the slicer with "Enable Label objects". It would be good if you see a screen with the names of the individual parts or as under Octoprint in "CODE VIEWER" an overview. https://github.com/paukstelis/OctoPrint-Cancelobject

hessius, 2019-10-13:
I dunno if possible but in my mind it would be better to use the Exclude region plugin: https://plugins.octoprint.org/plugins/excluderegion/ as that doesn't require slicer changes. Might be harder to actually implement though.

UnchartedBull, 2019-10-13:
Both plugins do have an API. However I would need to include some SVG drawing for Exclude Region, but this whole project was about learning something new. I might use cancel-objects for v2.0.0 and include exclude regions once the release is out of the door.

If both plugins are implemented the user could also choose which one should be used based on personal preference.