UnchartedBull/OctoDash #164: Deleted file still shown in File List

UnchartedBull, 2019-09-30:
Deleted Files are still being returned by the API and therefore shown in the UI. Confirmed with Postman, as deleted file is included in file list. File is gone in Octoprint UI File List immediately.

hessius, 2019-10-18:
I've noticed the same issue with the file manager plugin (see screenshot).

Shouldn't it be possible to match the path to exclude anything containing "trashe~"

[image]

UnchartedBull, 2019-10-18:
Oh that's actually quite intereseting. I never bothered to look at the full path for a deleted file and was just wondering why they don't show up. If they always include the trashe~ prefix it will be very easy to exclude them. Thanks for the tip!

hessius, 2019-10-19:
It also looks like the filenames all end with ~1 (potentially ~n) which should also be an easy regex filter

UnchartedBull, 2020-02-06:
Coming back to this. I just tried it on my system and I can't reproduce this issue, at least not in your way. If i delete a file via the UI it will still be shown in OctoDash, that's due to the fact that the API doesn't get updated though. As you can see in the screenshot the path doesn't change to `trashe*`. This is related to #400 as they both narrow down to the API not updating, even though the OctoPrint website is updating. I will create an issue in the OctoPrint repository tomorrow to hopefully get this fixed soonish. Once the API is updating correctly the files should be removed from OctoDash as well. I leave this issue open, just in case the then working api uses the `trashe*` path.

Screenshot of a file that was deleted:
[image]

Also I can't seem to find the trashe folder on my system. Is this maybe something that gets added by the plugin you mentioned? At least on my system there is no trace of a trashe folder:
[image]

UnchartedBull, 2020-02-06:
I just found this regarding the trash folder: https://github.com/foosel/OctoPrint/issues/1406. Are the files you are mentioning on the SD Card by any chance?

UnchartedBull, 2020-02-07:
Closing this in favour of #400 and https://github.com/foosel/OctoPrint/issues/3446

closed 2020-02-07 by UnchartedBull