UnchartedBull/OctoDash #167: Make arm64 compatible

purist69, 2019-10-01:
I dont use an RPi for my octoprint server, I use an arm64 device for performance purposes with a manual install of octoprint.  it would be nice if the installer supported arm64 as well.

Thanks.

UnchartedBull, 2019-10-01:
Hi this currently is not possible, because OctoDash relies on the BCM2835 library, which is only available for the Pi for temperature reading. It doesn't work good though so I'll probably will change that in the next minor version to use a Octoprint Plugin for that. So it probably will come but it takes a few days for me to change that.

UnchartedBull, 2019-10-01:
If you want to help out:

A super simple OctoPrint Plugin that reads a Temperature Sensor (DHT22 or DHT11) and just offers the values via an API would be super helpful!

UnchartedBull, 2019-10-08:
https://github.com/UnchartedBull/OctoDash/releases/download/v1.3.1/octodash_1.3.1_arm64.deb here you go. If you want to use the script just change every `armv7l` to `arm64`.

closed 2019-10-08 by UnchartedBull