UnchartedBull/OctoDash #180: unable to open octodash via VNC

wieltar, 2019-10-04:
I have an old s7 laying around and wanted to use it with the raspberry pi and vnc to connect to octodash, i just ran the install script and installed vnc. The connection is successful however i get the "unable to show desktop" message.

Is there some step i missed in the setup? my initial thought was that it would work the same as a physical lcd connected to the pi. It could very well be this isn't possible however i missed some documentation stating how i could connect a display

UnchartedBull, 2019-10-04:
It should be possible to view OctoDash via VNC. I don't know if the Pi creates the Desktop on boot even if there is no HDMI cable plugged in. Do you see anything in the VNC Viewer (like a black screen with a small white rectangle at the top right)?.

I don't use VNC Viewer therefore I might not be able to help here. I would probably install the normal Raspbian and get VNC working there (so you see the desktop on your phone) and then just install the no-autostart option and do that bit yourself.

I might also be able to offer OctoDash as a native Android App, which would make the whole VNC part obsolete, but that might take a couple of weeks ...

wieltar, 2019-10-04:
https://imgur.com/a/vhwuhAA

I've added a photo of the output on imgur.

I haven't seen the rainbow screen or the rectangle in the corner. I haven't tried installing the desktop first en then the manual part, I could do that in a bit though, do you have any special way of removing the plugin?

I'll try reinstalling it the manual way

UnchartedBull, 2019-10-04:
Okay the issue you are having is that there is no Desktop present, therefore VNC can't display anything. Your issue most likely lies within your configuration. You can remove octodash with `sudo dpkg --remove octodash`.

UnchartedBull, 2019-10-08:
I'm going to close this here so it doesn't show up in the tasks that needs to be done before the release. If you have any more issues just open a new one. Hope you were able to fix your issue.

You can also continue to comment in this issue :). Thanks!

closed 2019-10-08 by UnchartedBull

UnchartedBull, 2019-10-08:
There is now a native android app available: https://github.com/UnchartedBull/OctoDash/releases/download/v1.3.1/octodash-1.3.1.apk. I don't have an Android smartphone so I can't try it. Would be great if you could provide feedback.

yet-another-average-joe, 2021-01-19:
I know it is an old one !

Just install XTerm (terminal for X) : sudo apt install xterm
If you need the framebuffer to be displayed as well :

sudo nano /root/.vnc/config.d/vncserver-x11

add these lines :

CaptuteTech=raspi

CaptureTech=raspi gives gives access to the frame buffer, so VNC can also display non-X screens.
if you don't want the connection to be closed after some time, add :

IdleTimeout=0

But you have to know that you will probably get a very ennoying logging dialog from VNC from time to time. Probably a bug, found no reproductible pattern...

[image]

stefan-sherwood, 2024-12-17:
@yet-another-average-joe This worked great for me, thanks! I wish there was a way to suppress then VNC dialog on boot though as it appears in front of OctoDash and I have to choose "Hide" from a tiny menu every time.