UnchartedBull/OctoDash #191: Uninstall

TheNore, 2019-10-09:
I need to remove this dashboard and reinstall. I tried to upgrade from 1.2 to 1.3 and i get a "your config is invalid" error

UnchartedBull, 2019-10-09:
Did you upgrade to 1.3.0 or 1.3.1?

You have now two options:

- Fix the config (located at ~/.config/octodash/config.json and fix the errors
- Uninstall with sudo dpkg --remove octodash and reinstall

If you tell me the errors I can tell you how to fix them :)

TheNore, 2019-10-09:
1.3.1 sorry

UnchartedBull, 2019-10-09:
Ok I‘ll have a look at that in a minute. You have left the temperature field empty I guess?

TheNore, 2019-10-09:
yes, thank you

UnchartedBull, 2019-10-09:
Ok I re-released 1.3.1 which should now fix the issue (hopefully). Please execute the update script again and check whether the error goes away. The release was kinda rushed yesterday. Sorry for that!

TheNore, 2019-10-09:
yep! correct "new config" windows pops up

UnchartedBull, 2019-10-09:
Great! Thanks for the feedback. I'm closing this here. If you encounter any more issues please open a new one :)

closed 2019-10-09 by UnchartedBull

floridaservices, 2020-04-08:
OctoDash is new to me and I came across a similar problem tonight, I was allowed to remove an integer in a field and save but then  I was stuck with the "Your Config is Invalid screen".  Deleting the config.json will revert OctoDash back to new install status.  No need to uninstall.  This is to help future stuck newbs..