UnchartedBull/OctoDash #197: Installation error related to virtual environment

hessius, 2019-10-11:
Trying to install OctoDash using the install script I ran into some issues, I've tried going through the steps manually but still seeing errors.

Early on in the process I'm getting
`Installing OctoPrint Plugins
bash: line 7: /home/pi/OctoPrint/venv/bin/pip: No such file or directory
bash: line 8: /home/pi/OctoPrint/venv/bin/pip: No such file or directory
bash: line 9: /home/pi/OctoPrint/venv/bin/pip: No such file or directory
bash: line 12: /home/pi/OctoPrint/venv/bin/pip: No such file or directory`

And as the error message states there is no OctoPrint/venv folder (let alone venv/bin/pip).
There is however a /home/pi/oprint/bin/pip folder
I've used the official raspberry-pi image of octoprint, has the venv location changed?

Later on in the process I did get more error messages, probably related to the one above but attaching for the sake of completeness

`

19 packages can be upgraded. Run 'apt list --upgradable' to see them.
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 

chmod: cannot access '/usr/lib/xorg/Xorg': No such file or directory

dpkg: dependency problems prevent configuration of octodash:
 octodash depends on libgtk-3-0; however:
  Package libgtk-3-0:armhf is not configured yet.
 octodash depends on libnotify4; however:
  Package libnotify4:armhf is not configured yet.
 octodash depends on libnss3; however:
  Package libnss3:armhf is not configured yet.
 octodash depends on xdg-utils; however:
  Package xdg-utils is not configured yet.
 octodash depends on libappindicator3-1; however:
  Package libappindicator3-1:armhf is not configured yet.
 octodash depends on libsecret-1-0; however:
  Package libsecret-1-0:armhf is not configured yet.

dpkg: error processing package octodash (--install):
 dependency problems - leaving unconfigured
Processing triggers for hicolor-icon-theme (0.15-1) ...
Processing triggers for mime-support (3.60) ...
Errors were encountered while processing:
 octodash
`

I've installed Xorg and other dependencies manually and these issues do not occur when running the install script again now.

hessius, 2019-10-11:
Creating a symlink seems to have fixed the warnings when setting up

ln -s /home/pi/oprint /home/pi/OctoPrint/venv

Nothing happens on boot however, and running the command "octodash" from the command line produces the following message:

> : getrandom indicates that the entropy pool has not been initialized. Rather than continue with poor entropy, this process will block until entropy is available.

hessius, 2019-10-11:
Here’s the screen
[image]

UnchartedBull, 2019-10-11:
19 packages can be upgraded. Run 'apt list --upgradable' to see them.
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.

That's a big problem because you didn't install anything at all. Your package manager (dpkg) currently has some issue. Just run `sudo dpkg --configure -a` and they (hopefully) go away. After that you can run the install script again.

Did you rename the user to octopi or is that the default username for an OctoPi installation? That's the reason the script can't find your venv, because it looks in a folder that is for the user pi ... If octopis default user is octopi i would need to change that. If you wait a few minutes I can fix the install script so they use the username of the current user rather than pi.

hessius, 2019-10-11:
Yeah, I figured, I ran the dpkg command and was able to upgrade everything and after that ran the script again, that produced the result above. Sorry for not mentioning it in my first post. 
After doing that and the symlink the install script ran fine without any error messages, I just had to remove the extra lines from the bashrc file (as the lines had been added twice).

I did not rename the user.
I’ve run the install script over ssh, connected as user pi, however the screen looks to be expecting user octopi...

UnchartedBull, 2019-10-11:
Ok the issue now is, that all the startup codes are created for the pi user, not for the octopi user. According to your screenshot the octopi user is the one that is selected at default.

Do you have the option to login with octopi user?

hessius, 2019-10-11:
Hmm, I can certainly try, I don’t know the password for that user though, the documentation states that pi is the default user for ssh, and I haven’t found any documentation on the octopi user.
Is it a good idea to try to run passwd as root to try and edit the password or could I potentially mess everything up?

hessius, 2019-10-11:
(Or would it be possible to add the octodash process to the startup of the octopi user without access to the account using su?)

UnchartedBull, 2019-10-11:
I had a look at the OctoPi installation and it seems like they have a kind of weird setup. User pi is the default ssh user, but it seems like the whole OctoPrint installation and screen runs with the octopi user. 

You don't need the password to login though as long as you have sudo rights (which you should have). Could you try :

`sudo su - octopi` and enter your sudo password?

UnchartedBull, 2019-10-11:
If that doesn't work we can add everything to the octopi installation, but that is a litte more complicated. So try with `sudo su` first, as this should just need the sudo password and not the octopi password. You can then confirm who you are with `whomai`

hessius, 2019-10-11:
Yeah, pi has sudo rights.

I just headed out but will try tonight.

If I can run sudo su - octopi successfully, what’s the next step - Running the whole installation script again or just the xinitrc and bashrc changes?

I’ll try later on tonight

UnchartedBull, 2019-10-11:
You can run the whole thing again. dpkg is intelligent enough so it doesn't install the same thing twice. If that works, octodash should work normally after a restart.

If you can't loing with `sudo su - octopi` you can basically do everything from the script from line 27 on but replace each occurence of `~` with /home/octopi and then that should hopefully do the trick.

Please keep me updated once you've done that so I can update the README :)

spiff72, 2019-10-12:
Just commenting here to follow along and ask question...
I got a similar error during the install where the venv items near the beginning of the script fail, but I am installing under user 'pi' .  The trouble I see is that there is no venv folder under /home/pi/OctoPrint/ when I look at that directory. Is my OctoPrint install worked if this is the case? I honestly can't remember how I set it up (from a pre-made image or on top of raspbian). Needless to say the install is quite old, so perhaps the installation method for OctoPrint changed between now and then? I don't have any issues installing plugins through the web interface.

When I attempted to launch octodash, there was an error about not being able to run chrome sandboxed.

spiff72, 2019-10-12:
I also have been keeping OctoPrint up to date with its self update function.

hessius, 2019-10-12:
Ok, sudo su - octopi gives the following message:

> No passwd entry for user 'octopi'

So, I could of course run adduser octopi but I'm afraid I might screw something up... Any other ideas?

UnchartedBull, 2019-10-12:
@hessius - no don‘t add that user. Could you try replacing all the ~ in the Script with ‚/home/ocotpi‘, from line 27 on.

UnchartedBull, 2019-10-12:
@spiff72 if you also use OctoPi your venv probably is called oprint. I have to update the script once we get hessius up and running so I know what to do.

Could you be a bit more specific about the chrome sandbox error? Is that still valid or has it gone away?

hessius, 2019-10-12:
@UnchartedBull you can see the modifications I made here: https://github.com/hessius/OctoDash/blob/master/scripts/install.sh
The script fails as there isn't an octopi folder in /home/
```
bash: line 28: /home/octopi/.xinitrc: No such file or directory
bash: line 39: /home/octopi/.bashrc: No such file or directory
Setting Permission
chmod: cannot access '/home/octopi/.xinitrc': No such file or directory
```

UnchartedBull, 2019-10-12:
Which folders are listed in the home directory?

hessius, 2019-10-12:
only pi

hessius, 2019-10-12:
So maybe there's a way to run octodash as pi and take over the display? 
Running octodash generates the error message

> Gtk-WARNING **: cannot open display:

I know next to nothing about this (never attached a monitor to a pi previously) but that generates a number of suggested solutions when googling (e.g. https://www.linuxquestions.org/questions/debian-26/gtk-warning-**-cannot-open-display-0-0-a-807450/ )

UnchartedBull, 2019-10-12:
0.0 shouldn't work as a display identifier. export DISPLAY=:0 is the standard display for a Raspberry Pis Monitor. Could you try that and see if Pi has access to the display?

hessius, 2019-10-12:
Unfortunately no change:

> pi@octopi:~ $ export DISPLAY=:0
> pi@octopi:~ $ octodash
> 
> (octodash:817): Gtk-WARNING **: cannot open display: :0

UnchartedBull, 2019-10-12:
Wait a  second. I just had a look at your screen shot again. Can it be that octopi is only the name of your installation?

It looks like your not logged into the system yet. Could you try connecting a keyboard to your raspberry and provide your username (pi) and the password, so just enter pi where it says `octopi login:` and after that the system will ask for your password. That should log you into the machine and probably start octodash.

hessius, 2019-10-12:
You seem to be right... Octopi seems to be the installation, pi the user.

AND IT WORKS

Since I don't have an external keyboard (I tried attaching an old media remote but couldn't get it to work either) I instead configured the pi to automatically boot into console without requiring login using Raspi-config (for anyone else following along: https://www.opentechguides.com/how-to/article/raspberry-pi/134/raspbian-jessie-autologin.html )

After rebooting and automatically login I can see rat poison loading, but after that the screen remains black for a while, I thought it was stuck but went back to checking on it... I just now realised that while I had rotated the display, the touch layer wasn't rotated... That's not your problem however.
Thank you so much for your help - After I've got touch working and have set everything up I'll report back!

hessius, 2019-10-12:
Ok it's almost working now!

I don't know if this is an issue related to the setup and difference in location of things or if this is completely unrelated.

I can click through the on screen setup, but since I don't have a keyboard attached it can't connect.

Adding an edited version of the sample config.json to ~/.config/octodash/ does not seem to have any effect.
note: there was no file there before I added it, the directory is not empty however:

> blob_storage  Cache  Code Cache  config.json  Cookies  Cookies-journal  Network Persistent State

On rebooting I still get the initial setup screen, I've double checked and the config file is intact.
Any ideas?

UnchartedBull, 2019-10-12:
Does that screen say "Hey there!" or "Sorry for bothering you again ..."? If its the latter that just means your config is invalid, which is to be expected since I didn't update the sample config after the laster update (ups). If you replace the `"temperatureSensor": null` with 
```      
"temperatureSensor": {
    "ambient": null,
    "filament1": null,
    "filament2": null
},
``` it should work I think. I'm going to update the sample config as well

hessius, 2019-10-12:
No, it was the initial setup again.

I could get through the setup by running the xdotool type command (that was a really handy tool!)

Now trying to put it through its paces now. I've had one issue, not sure if octoprint or octodash related where it got stuck on "cancelling"

UnchartedBull, 2019-10-12:
Those are some helpful links. I'll make sure to persist those in the wiki/README. Thanks!

The issue with cancelling is often related to OctoPrint. OctoDash is just calling the API. Sometimes OctoPrint takes it time to cancel based on the GCode that has been sent to the printer. If you're heating the bed you have to wait until the bed is heated for OctoPrint to cancel. But normally after 1 or 2 minutes the cancellation is successful.

I'm going to close this here. If you have any more issues please open a new one :)

closed 2019-10-12 by UnchartedBull

hessius, 2019-10-12:
Great, as mentioned previously @UnchartedBull huge THANK YOU for your time! Hopefully this can be useful for others as well. Great job on OctoDash! (You should really try to get people engaged on the octoprint forums, they're gonna love this!)

spiff72, 2019-10-12:
I think you are correct about the oprint venv.  I just read about that last night when I looked at some of the octoprint documentation.

I will get more details on the sandbox error momentarily.  I assumed that it was related to the failures in the install script.

spiff72, 2019-10-12:
should i start a new issue since tthis one closed?

UnchartedBull, 2019-10-12:
@hessius I want to finish it first there are still some parts that are missing, after that I'm taking this to the forums :) But spread the word if you like it 👍 

@spiff72 Yes, please. I updated the install script so it should work with OctoPi as well. If the sandbox issue still occurs after that please open a new issue and I'll be happy to help :)

spiff72, 2019-10-12:
Issue opened:
https://github.com/UnchartedBull/OctoDash/issues/198

Thanks!