UnchartedBull/OctoDash #198: Errors on launch - possible missing dependencies?

spiff72, 2019-10-12:
When trying to run octodash, I get several errors - see attached screen shot.

One of them is a sandbox error, but there are a few other errors relating to ratpoison, libdbus, and g2d_23. 

I am using this screen - purchased about 3 years ago, and dug out of a box of spare parts to see if this octodash setup would work.  I have been struggling getting the touchscreen calibration to work on this (xinput-calibrator), as the settings when i save them in the /usr/share/X11/xorg.conf.d/99-calibration.conf file don't seem to "take" (presses at the center seem accurate, but presses toward the outside of the screen are registered about 1cm closer to the center of the screen than where i touched).

[image]

spiff72, 2019-10-12:
Oops - forgot to link the screen:
https://www.amazon.com/gp/product/B013JECYF2

spiff72, 2019-10-12:
Oh boy - I just did a sudo apt-get dist-upgrade and it seems to have fixed the problem!  

Now I need to figure out this calibration problem so i can accurately touch the screen icons. :-)

UnchartedBull, 2019-10-12:
Ok so the issue went away? Upgrading first is always a good idea! :) But I‘ve also wasted a lot of times on errors that are fixed with a simple upgrade.

spiff72, 2019-10-12:
I had neglected to do the dist-upgrade (only did upgrade).  I will close this one out.  The script changes fixed the issues with oprint.
I also poked around a bit in the interface and found that I needed to change the 5000 port to be blank for it to work.  
Two questions:
Is there a way to sort the files list by date uploaded (so the most recent are at the top)?
Does the screen ever go blank (like a screen saver)?

closed 2019-10-12 by spiff72

UnchartedBull, 2019-10-12:
Did you configure your OctoPrint to run on port 80? 

Currently there is no way to sort the files (they are just sorted by type and name). Maybe in a future version this will be support :).

Screen does never go blank no. But this can be setup with Raspbian I guess.

spiff72, 2019-10-12:
It does seem to be port 80.  I tried [ip address]:80/api/version and [ip address]/api/version and both report the version info.

I set this all up several years ago, so I don't remember the specifics.

spiff72, 2019-10-12:
One other question - how do you exit the octodash program?  I would like to stop it so I can launch the desktop and re-try the screen calibration to see if my dist-upgrade fixes my issue with calibration.

UnchartedBull, 2019-10-12:
You can list all programs running and grep octodash from that. Than just kill the process. Sorry there is no easier way :/ Maybe ratpoison does have a command as well but I‘ve never really searched for that ...

spiff72, 2019-10-12:
I ended up just commenting out the various lines that were added to .bashrc and .xinitrc so it doesn't start at boot.
I am still troubleshooting the touchscreen calibration because I want to rule out errant touches causing some other odd issues with the octodash program.  I am getting a lot of error message popups as I navigate the interface, and they appear to be related to the octoprint server restarting.  I suspect this because when i have the octoprint web interface open on my other computer, it pops up the "reload" dialog asking to reload the server - and the printer has then disconnected.

I might post another issue if I can't get this one resolved - but the touchscreen calibration has me running in circles.

UnchartedBull, 2019-10-12:
Did the whole restart thing happened before you installed OctoDash? Normally OctoDash shouldn't restart OctoPrint. The only thing I can think of is, that maybe your touchscreen registers 1 touch as multiple ones and therefore octodash sends multiple requests to OctoPrint, which might overload the system. But it's just a guess in the blue here.

spiff72, 2019-10-12:
No the restarts only happen with Octodash.  I am still fighting with the calibration here - I ended up starting over with a new copy of my current octoprint server and this time I have upgraded the distro to stretch to see if that helped with the calibration.  It hasn't.  Grrr.

So I haven't yet attempted to reinstall octodash until i get that resolved (if it is resolvable).  No matter what I do with the calibration - it just doesn't seem to accept it.  I think the xinput calibrator is generating odd numbers (very large - 60000 or more for the values) - and this doesn't seem to align with others results as I am searching this online.

UnchartedBull, 2019-10-13:
Damn that issue sound like a real pain. If you manage to fix this and the issue with OctoPrint restarting still persists you may want to try to lower your API interval, if you have set that to a too short amount of time you may run into issues.

spiff72, 2019-10-14:
Yeah - 

I have given up for now.  Until I find a reasonable touchscreen I have stopped my efforts on this one.  I was going to attempt modeling an enclosure for the pi and the display, but I held off until I was sure it was working. 

I will keep an eye on this project - I think it has a lot of potential.  I will probably look for another screen - maybe a capacitive one this time!

UnchartedBull, 2019-10-14:
Ok please create a new issue once you've come back and if you experience the same issue :)

spiff72, 2019-10-14:
Will do - thanks again for your help.

I am looking at some other screens - would a 1024x600 resolution (or other higher resolutions) work with this, or should I stick with 800x480?

UnchartedBull, 2019-10-14:
I've tried making it scalable to almost every resolution. Just try to stick with an 5:3 aspect ration roughly. The screen should work, also in some screens everything might be a little bit crunched/stretched at the bottom. Shouldn't be a big problem though as 1024x600 is only slightly off 5:3

This is what it looks like on 1024x600

[image]

UnchartedBull, 2019-10-14:
I can try and get some more screenshots once I get home if you like

spiff72, 2019-10-14:
No that is all i need to know.  I keep seeing some of these slightly higher res screens and wondering if they would work.  Higher res might be nice for just showing the Octoprint TouchUI too...

UnchartedBull, 2019-10-14:
I mean OctoDash will look better on higher res displays, because everything just seems sharper. I've never tried TouchUI, because I don't like the look, that's the whole reason OctoDash exists :D. But that is personal preference :)