UnchartedBull/OctoDash #199: compatibility problem with waveshare 3.5ich ips  lcd

mvoss96, 2019-10-12:
Hi, first of all thank you for your project.
I was trying to install octodash to use it with an wavesghare 3.5inch ips touchscreen (https://www.waveshare.com/wiki/3.5inch_HDMI_LCD)
However it seems like there are some compatibility problems:

I startet with octopi stretch lite then installed octodash which woked perfectly however when i installed the waveshare touchscreen driver octodash does no longer work.

the commands for the waveshare driver wer:
`git clone https://github.com/waveshare/LCD-show.git
cd LCD-show/
./LCD35-HDMI-480x320-show`

now i only get the console screen and it says: 
 ```
fatal server error
(EE) no screens found(EE)
```

the content of the Xorg.0.log file is:
```
[    14.069]
X.Org X Server 1.19.2
Release Date: 2017-03-02
[    14.072] X Protocol Version 11, Revision 0
[    14.073] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
[    14.075] Current Operating System: Linux octopi 4.19.66-v7+ #1253 SMP Thu A$
[    14.075] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.f$
[    14.084] Build Date: 18 October 2017  04:55:30PM
[    14.086] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support$
[    14.087] Current version of pixman: 0.34.0
[    14.089]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    14.089] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.099] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Oct 12 23:56:39 20$
[    14.106] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.115] (==) No Layout section.  Using the first Screen section.
[    14.115] (==) No screen section available. Using defaults.
[    14.115] (**) |-->Screen "Default Screen Section" (0)
[    14.115] (**) |   |-->Monitor "<default monitor>"
[    14.121] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    14.121] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    14.121] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    14.121] (==) Automatically adding devices
[    14.121] (==) Automatically enabling devices
[    14.121] (==) Automatically adding GPU devices
[    14.127] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.128] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    14.128]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    14.129]    Entry deleted from font path.
[    14.129] (==) FontPath set to:
        built-ins
[    14.129] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.129] (II) The server relies on udev to provide the list of input device$
        If no devices become available, reconfigure udev or disable AutoAddDevi$
[    14.131] (II) Loader magic: 0x1fbf40
```

alos it seems like sudo nano /usr/share/X11/xorg.conf.d is empty if this is relevant

UnchartedBull, 2019-10-13:
Hmmm I don't have a waveshare display so I can't really test anything. The error indicates that you have some issues with your screen rather with OctoDash though.

Some things you might try;

- Could you start ratpoison / octodash manually?
- Open an issue in the waveshare repository and ask them whats going on, I think the issue lays on their side
- Rerunning the installation script

mvoss96, 2019-10-14:
Hi, i have solved the issue. The waveshare script appears to be truely broken. i copied all settings manualy in the corresponding files and now the touchscreen works flawlessly.

The UI really is very beatiful.
However i would have one suggestion: Maybe there could be a bit more feedback on touchscreen input so you see when a touch is registered.

thank you for your help

UnchartedBull, 2019-10-14:
Great that this is working now. Going to close this issue for now.

would you be so kind and open a new issue for the touchscreen feedback? I've only seen in the control screen that the feedback is quite little ... Please open a new issue and elaborate a bit further which screens you would like to improve.

closed 2019-10-14 by UnchartedBull

markostamcar, 2020-04-18:
I just wanted to add (it might help someone) how I fixed a problem with my Waveshare 12287 3.5inch RPi LCD (B):

I installed a fresh OctoPi 0.17.0 and OctoDash using the script and enabled Console auto-login however X would fail with the error message "Failed to load module "fbturbo" (module does not exist, 0)".

The solution is simple :) Just install "sudo apt install xserver-xorg-video-fbturbo"

BTW @UnchartedBull, thanks for the beautiful UI, it looks great on my Ender 3 Pro :) https://imgur.com/a/EzcksY9