UnchartedBull/OctoDash #200: Turn off screen when idle

hessius, 2019-10-13:
Feature/enhancement request:

I want to reduce power consumption for my printer and would like the screen to turn off when idle.

To just "black out" the screen you could easily use 
`xset dpms force off`
This makes the screen black but still responsive to touch, i.e. easily woken up by touching the screen.
IMO that's good enough for a first implementation, when idle for 15 minutes (configurable) black out screen.

Ideally I would want the backlight to turn off as well but I have yet to figure out how to do that (I've seen some solutions specifically for the official rPi screen but for a project like this you would need something universal)

As a proof of concept I've now added the command to my PSU-control setup (which controls an Ikea Trådfri smart socket that supplies power to the printer and lights around it), but would love a solution that isn't directly connected to the PSUcontrol plugin as the screen could then be used to power on the printer (I'm posting separate issues around these features)

UnchartedBull, 2019-10-13:
Will be useful I think, i'll have a look if electron allows me to execute console commands (which it should) and how I can get everything back up again.

UnchartedBull, 2019-10-15:
This will be done via `xset dpms force standby` (this also turns the official RPi display completely off). Could you please check if this works for you as well?

UnchartedBull, 2019-10-15:
Ok I reopen this issue because I haven't tested it on the Raspberry. If you set `turnScreenOffSleep` to true the screen should turn off after 5 Minutes (only in standby screen, which gets activated once the OctoPrint connection is closed). 

Will close this again once testing is done (before next release)

hessius, 2019-10-16:
`xset dpms force standby` seems identical to `xset dpms force off` on my device. Screen is black but backlight is still on

UnchartedBull, 2019-10-16:
I think most of the non-official Raspberry Screens don‘t have the backlight turn off implemented :/ can you turn off the screen with any command?

hessius, 2019-10-16:
Unsure, I'll keep looking!

UnchartedBull, 2019-10-23:
I wanted to do some testing if everything works when I'm calling xset via electron. Which proves to be working just fine. I'll close this here.

**Important:** You need to enable `turnScreenOffSleep` in the `~/.config/octodash/config.json` file. It is by default (will be included in the  settings menu once that is done).

closed 2019-10-23 by UnchartedBull

ShagoY, 2019-12-07:
Hi :)

There is a possibility to turn the screen ON after using turnScreenOffSleep?

My Octoprint server runs continuously because it turns my smart plug on/off. But unless I make a mistake in my configuration, the screen and Octodash don't seem to reconnect or turn on the screen automatically.

My printer is in a fully sealed enclosure and I mainly use Octodash to know the status of my prints. So being able to see at a glance if it is on and/or being printed would be great!

UnchartedBull, 2019-12-07:
Currently you have to tap the screen once to wake up OctoDash. Does that work for you? If you want to turn the display back on, without touching the screen, that does not work yet ...

ShagoY, 2019-12-08:
Not yet but planned ?

UnchartedBull, 2019-12-08:
If you open a new issue for that, it will be planned at some point ;)

PR #213 (2019-10-15): Enable standby mode
Enable standby mode (automatic reconnect on wake up) and PSU Control.

Standby mode is still funky, because I can't kill Singleton Services, which leads to some issues. These are going to stay until after the v2 release, as it works 90% of the time.

Closes #200 #201  and #202