UnchartedBull/OctoDash #201: Control PSU from OctoDash

hessius, 2019-10-13:
As I mentioned in #200 I would love the ability to control the PSU-control plugin through the OctoDash screen. 

Without knowing any of the details: I'm using the app OctoPod which can control the plugin so there is some sort of API.
It's also possible to set up PSU-control to respond to Gcode commands, (defaults to M80/M81), so this could probably be added to the custom controls? Though without having tried it yet it looks like the power on Gcode command would not be usable since OctoDash throws a 409 error continuously when the printer is offline?

UnchartedBull, 2019-10-13:
Yes if the printer is offline OctoDash can't send anything, because everything will go through OctoPrint which will throw the 409 error. But thats a good idea and fits really well into the custom actions. Also kill would make kind of more sense then. So that feature probably is  coming soon :)

Could you link the plugin so I can have a look?

hessius, 2019-10-14:
Here’s the plugin: https://github.com/kantlivelong/OctoPrint-PSUControl
Here’s a write up I made on how to set it up with ikea Trådfri smart outlets: https://community.octoprint.org/t/psu-control-with-ikea-tradfri-and-no-additional-hw/12536

UnchartedBull, 2019-10-14:
It does have an API so the plugin can be integrated quite easily :)

hessius, 2019-10-15:
Small update; enabling the save connection settings and the auto connect on server startup in conjunction with this plugin enables auto connection after powering on the psu: https://github.com/markwal/OctoPrint-PortLister

UnchartedBull, 2019-10-15:
I've almost finished the sleeping screen, which also attempts to connect to the Printer (just needs some testing and some minor tweaks). And because I don't use the WebInterface, but rely on the API it hopefully should work just fine. Maybe if this doesn't work out I'll have a look at the plugin

closed 2019-10-15 by UnchartedBull