UnchartedBull/OctoDash #202: Improve offline response & Connect to Printer

hessius, 2019-10-13:
As mentioned in #20 and #201 I would love to have an improved error handling for offline printers.
As I turn off my printer automatically after finishing a print it's not uncommon that the printer is offline. Even after turning on the printer it's not connected (and I haven't been able to get it to automatically connect either).

So I would like an error message that isn't as obtrusive as the current one, but maybe displays as a red bottom bar (instead of a modal that needs to be closed). Ideally this message would have a proposed action like "connect"

UnchartedBull, 2019-10-13:
Yes that is one of the points that need to be addressed some time soon. I'll have a look at how this could be done. One thing to keep in mind is the memory leak prone console. As Chrome keeps all errors in the console forever I think.

hessius, 2019-10-14:
I did not know that!
And running 
`console.clear() `
Doesn’t clear errors either?

(Documentation https://developer.mozilla.org/en-US/docs/Web/API/Console/clear)

UnchartedBull, 2019-10-14:
I have to do some digging around that, but it seems like this function can be used in Chrome. Big Question is, whether all the pointers to the variables are cleared as well, so they can be garbage collected. I'll do some research and think of something that could be integrated ...

UnchartedBull, 2019-10-15:
It is kind off working with some hickups here and there. I now want to gradually introduce a future stop for v2, because otherwise it will never get finished. #214 already addresses the issue and will be worked on, once v2 is finished.

closed 2019-10-15 by UnchartedBull