UnchartedBull/OctoDash #204: Multiple Cameras

CamoJackson, 2019-10-13:
Please, add support for multiple camera streams into full screen and tab view

hessius, 2019-10-13:
I think you might be on the wrong repo, this repository is not for the dashboard plugin but rather a dashboard to be displayed on an external screen.

UnchartedBull, 2019-10-13:
OctoDash already supports Camera Views actually (kind off). You can open an iFrame with Custom Actions: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions which could show your webcam stream (see command [WEB]). Is that sufficient?

CamoJackson, 2019-10-13:
Sorry,  your right wrong repo

closed 2019-10-13 by CamoJackson

Swap-File, 2020-08-07:
I'd like to say that camera support is one of the primary reasons I switched to OctoDash.  It might be useful to make that ability more prominent.

UnchartedBull, 2020-08-10:
I never thought that so many people are looking for that feature. I'll make sure to make it more prominent :) Thanks for the info!