UnchartedBull/OctoDash #208: UI increase scroll bar contrast

hessius, 2019-10-14:
As title, a small ui change; I thought scrolling the file list was broken because I couldn’t understand the scroll bar, would be good to have a higher contrast element, maybe an outline/border with somewhat higher contrast would be both unobtrusive and clear enough to improve usability. 

As a second change maybe up and down scroll buttons for the scroll bar would be good as well (instead of only relying on dragging the scroll bar handle)

(Btw are you using some sort of ui framework or making it up as you go?)

UnchartedBull, 2019-10-14:
Currently there is only a visible thumb for the scrollbar because I don't like any background for scrollbars (similar to how mac and iPhones handle scrollbars). A outline or something might be worth a look though it may break the current workaround for resizing the scrollbar in height though ...

You normally should be also able to just swipe on the file list, this works at least for me. Some buttons are a good idea though as it can be tricky with a resistive touch to record a smooth sliding motion.

Not using a UI Framework, I've never really found one, that I liked 100% so I though for a small project like OctoDash doing my own stuff should be sufficient.

hessius, 2019-10-14:
Okay, makes sense with the framework (just wanted to know because if so I could poke around that)

I’m not able to swipe on the list to scroll. 

Are resistive touch screens able to send the tap hold event reliably?
There are some smooth scrolling libraries, you could bind a tap to a certain scroll length with easing and then bind tap hold to a continuous scroll based on how long the user holds? (Haven’t tried this myself)

hessius, 2019-10-14:
(I’ve previously used a framework called framework7 which both includes loads of ui components and a DOM manipulation library (which can be used separately, called DOM7) where I think this feature could be built using the built in scrolling functionality, this might all be overkill for this feature though.)

UnchartedBull, 2019-10-14:
Currently it's just a list with overflow so Chrome takes over the whole scrolling part. Don't know if I can change how Chrome handles that.

Oh ok if also take DOM Frameworks and stuff - I'm using Angular 8, which makes everything a lot easier (two databinding, etc.). I may have a look whether there is a component for that, but having two of these libraries running in parallel good lead to some funky behavior.

hessius, 2019-10-15:
Yeah I agree it might be overkill, you can however use just the ui component library and roll your own library for the Dom. They have pre made packages integrating vue and react but left angular a few versions ago.

It’s weird that I can’t drag to scroll on the list (that’s what I tried at first), ensuring that works would remove the need for higher contrast, maybe something can be done to improve/enable scrollability...

Maybe https://github.com/cubiq/iscroll could be a good fit, it’s a light weight library with good documentation but unfortunately not maintained as the author retired. That’s more of a work around than a straight up bug fix though.

UnchartedBull, 2019-10-15:
I have a look at how the scrolling could be improved and may come back to you for testing, if you don't mind :)

hessius, 2019-10-15:
Yeah absolutely!

UnchartedBull, 2019-10-17:
Ok after some debugging and tinkering around I think I found the issue:

Can it be that you only tried dragging on files? At least on my side the library I use to detect long presses seem to destroy the scroll event. Could you try and scroll to folder and start dragging there and confirm that this works for you?

hessius, 2019-10-18:
Yeah I've only scrolled on files since I haven't got folders. But just so that I understand, you want me to try dragging on folders instead of files to confirm it's the same behaviour that you found?

UnchartedBull, 2019-10-18:
You basically have to start scrolling on a folder and then can go over files and stuff. Just the start needs to be one a folder. But I‘m 99% certain, that this is the problem here. I already found a fix for that.

hessius, 2019-10-18:
Ok, great I recently purged all my old files, I'll add some back after work tonight and try scrolling starting on a folder

UnchartedBull, 2019-10-18:
Yes that will be great. I'll wait with the merge until you confirm that scrolling on starting on a folder does work :)

UnchartedBull, 2019-10-18:
I've also included a track for the scrollbar thumb, it looks a bit nicer and cleaner that way :) You can see a small part of it at the very bottom.

[image]

hessius, 2019-10-18:
I can confirm scrolling on folders is working

PR #233 (2019-10-18): Scrolling working on files again
Fixed an issue with hammerjs that caused a broken scrolling handling on files. closes #208.
closed 2019-10-18 by UnchartedBull