UnchartedBull/OctoDash #210: OctoDash dont read/show files

ARCH-3D, 2019-10-14:
Hello, 
When i touch in "files" dont show the files that i have to print. Appears a screen with a pacman "loading" but dont show anything else. What can i do? Thanks in advance

**General Information:**
 - Raspberry Pi
 - OctoPi
 - OctoDash Version [last update]

UnchartedBull, 2019-10-14:
Do you have lots of files? Are there any errors showing up or is it just stuck at the loading screen? How long have you waited?

It would be great if you could send me the output of the API. You can do this easily via Postman:

GET <octoprint-url>:<octoprint-port>/api/files and make sure to include your API Key in the Header Field `X-Api-Key`. If you need any more help to retrieve this item let me know. You can also send the output to my email (timon.gaebelein@icloud.com) if you don't want to share the output here. Thanks!

ARCH-3D, 2019-10-14:
Thanks for your answer UnchartedBull. Maybe 50 files. I waited more than 10 minutes, with no errors, just stuck at the loading screen.

UnchartedBull, 2019-10-14:
Ok then there probably went something wrong while trying to parse the files. I would need to get the output from that API in order for me to fix this problem, as I never faced something similar. Sorry.

ARCH-3D, 2019-10-14:
Ok Thanks. I sent you an email. You received?

UnchartedBull, 2019-10-15:
Yeah I've already replied :)

PR #230 (2019-10-17): Fixed multi material issue
This issue has been resolved, it was due to multiple materials being used (not including tool0), as well as a bug in the code that tried to access GCodeAnalysis for non GCode Files.

The displayed filament amount is now the combined amount of all materials used.

Closes #210.
closed 2019-10-17 by UnchartedBull