UnchartedBull/OctoDash #212: Improved touch callout suggestion

hessius, 2019-10-15:
As mentioned in the #199 issue en passant, touch callout could be improved throughout the UI. IMO opinion this could be said of any button pressed in the UI.

**Describe the solution you'd like**
I think a "ripple" effect as in Material design could be implemented quite easily and not break with the rest of very clean UI.

**Link other projects**
Several libraries exist to help implement a ripple effect:
Larger library: https://material.io/develop/web/components/ripples/
Smaller libraries:
Material-ripple effect [CODE](https://github.com/davinder17s/material-ripple-effect) [DEMO](https://davinder17s.github.io/material-ripple-effect/) 
rippleJS [CODE](https://github.com/samthor/rippleJS) [DEMO](https://samthor.github.io/rippleJS)

UnchartedBull, 2019-10-15:
Yes that is a very good idea. I would mainly use it for the control view, are there other places you can think of where this can be used?

UnchartedBull, 2019-10-15:
And the files view of course

hessius, 2019-10-15:
yeah anywhere there’s interactivity really

PR #249 (2019-10-23): Feature/touch callout
Closes #212
closed 2019-10-23 by UnchartedBull