UnchartedBull/OctoDash #214: Improve standby mode

UnchartedBull, 2019-10-15:
**Is your feature request related to a problem? Please describe.**
Standby mode continues to call the API, which is unnecessary and can lead to a Race Condition that causes one 409 Error to show up. Need to kill Angular Singleton Service for that somehow (maybe reload?)

spiff72, 2019-10-17:
Following up on this one - if there was any way to represent the "Connect" button that is present in the web interface I think that would be very nice addition to the feature set.  I have made some adjustments to the custom commands to suit my printer (Home doesn't do a bed level check, different temps for the preheat, etc.), but I would love to see a button that would let me connect to the printer after a reboot.

UnchartedBull, 2019-10-17:
That actually is included in the current revision of the standby mode. If OctoDash detects that the Printer is not connected it will go into sleep mode. Once you press the screen it sends a connect command to OctoPrint and will reenable itself. There is just no separate button ...

spiff72, 2019-10-17:
Ok good to know. I will check this out further. I (re)installed this after getting a new screen to use with it. I set it up on Tuesday so if things changed since then I am not up to date.

UnchartedBull, 2019-10-17:
It isn't released yet (just for my dev build). You can have a sneak peak at new features in the screenshots folder mostly though: https://github.com/UnchartedBull/OctoDash/blob/master/screenshots/sleeping.png ;). Will be included in the next release.

spiff72, 2019-10-19:
> That actually is included in the current revision of the standby mode. If OctoDash detects that the Printer is not connected it will go into sleep mode. Once you press the screen it sends a connect command to OctoPrint and will reenable itself. There is just no separate button ...

Out of curiosity - is there something going on behind the scenes that makes this able to connect to the printer in a case where the raspberry pi has just been rebooted?  I have NEVER seen this happen with my installation and printer (Prusa i3 MK3S).  I have the "automatically connect at server startup" item ticked in the connection section of Octoprint, but it never connects after a reboot - I always have to click the connect button to get the printer connected.

UnchartedBull, 2019-10-19:
It will show that message everytime OctoPrint is disconnected, so after a reboot as well if the autoconnect isn't working. Just tap the screen and it will connect to the Printer :)

spiff72, 2019-10-19:
I discovered that I needed to specify the actual port (not AUTO), save that setting, and now it does AUTO connect after a reboot.

UnchartedBull, 2019-10-19:
Great that you sorted that out! Going to leave this issue open, because there still is some stuff to do regarding standby :)

PR #378 (2020-01-22): Fix/standby improvements
Improve standby, still not perfect, but killing the singletons is currently out of scope for this project. Notifications are catched earlier now and the console is cleared on a reconnect, which should clean up the possible garbage.

Connection is checked before triggering a connect command to prevent restarting. Notification are muted until the connection is established, which should prevent random errors popping up.

PSUControl is also added natively now.

Closes #214 
Closes #368
Closes #371
closed 2020-01-22 by UnchartedBull