UnchartedBull/OctoDash #218: 409 CONFLICT error with OctoDash

WishmastR, 2019-10-16:
[image]

Hi

I've got a RPI3 running Octoprint with a 3.5" LCD screen attached to it.
I used to run Octoprint-tft on it without any issue. 

With OctoDash I managed to go through the setup process, using the API key related to the Octoprint account I use (with built-in login/password protection).

But in OctoDash the only thing that seems to work is the file browsing feature (for files in Octoprint), for everything else I've got an error 409 pop-up that keeps appearing (see screenshot).

Any idea on how to fix it?

Thanks

UnchartedBull, 2019-10-16:
Your printer probably isn‘t connected to OctoPrint. Please make sure (via the WebInterface) that your printer is connected to OctoPrint.

WishmastR, 2019-10-20:
Damn you're right, I feel stupid not having checked this first.
Human error mistake, it can obviously be closed

closed 2019-10-20 by WishmastR

Mearman, 2020-07-03:
@UnchartedBull Is there any reason why I might get 409 when the printer definitely is connected?

UnchartedBull, 2020-07-05:
Does OctoPrint show a connection in the top left corner? This error usually only shows up if the connection isn't working or you're trying to start an invalid action while printing, i.e. starting a new print.