UnchartedBull/OctoDash #225: Confirmation Dialog for some custom commands/actions?

spiff72, 2019-10-17:
**Is your feature request related to a problem? Please describe.**
Somewhat...
Some of the commands in the custom commands/actions are "dangerous" - e.g. Shutdown, Kill). 

**Describe the solution you'd like**
Could there be a simple confirmation dialog that could be implemented in cases where a command could be harmful to the system?  (Maybe just add a flag to the config file that adds the confirmation dialog for certain commands?)

UnchartedBull, 2019-10-17:
Great idea. I'll definitely make sure to include that.

I'm currently trying to do some kind of a Feature Stop for OctoDash v2 so it finally gets released. Therefore I'll move this issue to after the release of OctoDash. Sorry for the wait. It will be one of the first issues taken care off after the v2 release :)

spiff72, 2019-10-17:
No problem. Your responsiveness on this program is exceptional!

UnchartedBull, 2020-03-02:
Was introduced in 1.4.0, can be adjusted in the settings

closed 2020-03-02 by UnchartedBull