UnchartedBull/OctoDash #226: Add ability to "Sort by upload date (descending)" to files list

spiff72, 2019-10-17:
**Is your feature request related to a problem? Please describe.**
No - Simply a convenience feature

**Describe the solution you'd like**
As described in the title, adding "Sort by upload date (descending)" would be a FANTASTIC ease of use enhancement for starting a print on the most recently uploaded file.

UnchartedBull, 2019-10-17:
Great idea. I'm putting this in for the v2 release for now, I may need to move it later though.

spiff72, 2019-10-25:
Is there a way to "subscribe" to know when v2 is released?  I just had to re-image my octopi to get it up to the latest version of octopi, and I am thinking I will just wait into install octodash until the release is done (if it is coming soon).  Or is upgrading as easy as entering the terminal command on the main readme page?

UnchartedBull, 2019-10-26:
You can watch the repository, GitHub will then send you an email once a new release is ready. Upgrading OctoDash is also quite easy. Most of the time the configuration get's updated automatically, so there really is no hassle involved. If it can't be update automatically OctoDash will show you the screen, that is shown at the beginning, but all values will be prefilled, so this should be really easy.

With v1.3.2 OctoDash will also show you a small notification within the program to . let you know that an update is ready :)

UnchartedBull, 2019-10-26:
So yeah it is easy as executing that one liner from the README :)

spiff72, 2019-10-26:
Got it - thanks.  I didn't realize that there had been some point releases since I installed.  I went ahead and installed 1.3.2 now (after some fighting with the install script - I had to install manually because ratpoison and xorg stuff apparently didn't get installed by the script.  Probably operator error here...

PR #443 (2020-02-20): Allow sorting of files
Closes #226
closed 2020-02-20 by UnchartedBull