UnchartedBull/OctoDash #235: Bed Termperature not rendered correctly - possibly when temp is 100C or above

spiff72, 2019-10-19:
**Describe the bug**
Bed temperature display isn't rendered correctly.

**To Reproduce**
My display resolution is 1024 x 600 (7" screen).
Not 100% sure this is the cause, but I noticed this when attempting my first print with ASA, which has a bed temperature setting of 105 C.
**Expected behavior**
Temps of 100 C or more should render correctly.

**Screenshots**
Image is attached showing the issue.

**General Information:**
 - Hardware: Raspberry Pi 3B+, 1024 x 600 captouch screen
 - OS Info: Raspbian Buster
 - OctoDash Version: Latest Version?  Not sure where to check this

**Additional context**
[image]

spiff72, 2019-10-19:
Octoprint info:

OctoPrint 1.3.11 running on OctoPi 0.13.0

spiff72, 2019-10-19:
Strangely, after the second layer started (and the temp was increased to 110 C, the temp is rendered correctly.
[image]

spiff72, 2019-10-19:
One last comment for now.  I noticed when the bed temp sagged to 109 C the rendering issue came back (sorry - no pic).  So perhaps the rendering is incorrect when temp is between 100 and 109 C (inclusive)?

UnchartedBull, 2019-10-19:
No it’s just I never thought that the printed bed does never heat up to over 100C, that’s why I only tested til 99, which worked. I‘ll make sure to include the fix.

110 works because it is not as wide as 150. 101 for example will work as well.

spiff72, 2019-10-19:
Ah - that makes sense.

spiff72, 2019-10-27:
Is this closed as of 1.3.2, or is the fix coming in v2?  I think that I am running 1.3.2, but not positive (is there a way to check the version?)...and the rendering problem is still present on my 1024 x 600 display.  Layer counts are getting chopped off at the bottom too.

spiff72, 2019-10-27:
Here is a photo i took today.
[image]

UnchartedBull, 2019-10-28:
It has been fixed in v1.3.2 for 480x320. Haven‘t tested it on 1024x600. I‘ll make sure to include a fix in the next version :)

PR #237 (2019-10-20): Fix Layout getting funky, if bed temperature is over a 100 degree
closes #235
PR #287 (2019-11-04): Fixed on 1024x600
Closes #235
closed 2019-11-04 by UnchartedBull