UnchartedBull/OctoDash #236: Fan speed is not updated

mvoss96, 2019-10-19:
I noticed the following behavior:
when printing a gcode via octoprint the fan speed is correctly reported on the UI however if i change the fan speed via the printer lcd  (reprap full graphics smart controler) the fan speed in the octodash ui is not updated. Not sure if this is a bug or just a limitation of the gcode protocol.

UnchartedBull, 2019-10-19:
I can‘t fix that unfortunately. The printer does not return much information about its state. The Fan Speed is none of them. That’s why it can only be calculated based on the commands send to the printer. 

Going to close this here, once OctoPrint gets this figured out (I won‘t put much hope in it though), I‘ll update OctoDash.

closed 2019-10-19 by UnchartedBull

martijnvankekem, 2021-10-02:
Prusa printers have the GCODE M123 available to get the Fan PWM (0-255)

https://www.reprap.org/wiki/G-code#M123:_Tachometer_value_.28RepRap_.26_Prusa.29