UnchartedBull/OctoDash #256: Possible "interference" with PSUControl?

hessius, 2019-10-25:
**Describe the bug**
I've been using the OctoPrint-PSUControl plugin: https://github.com/kantlivelong/OctoPrint-PSUControl
To control the power of my printer using terminal commands.

While the PSUControl button still works, the automatic shutdown after a print has finished has stopped working.

The behaviour started some time after enabling OctoDash, I'm guessing that the plugin is sensing some sort of command being sent and therefore staying awake.

I'm however unable to find anything in the terminal output in OctoPrint that can corroborate that suspicion as I've added the only other GCODE than M105 (temperature report) that I could find: M117 (Set LCD Message) to the list of ignored commands but to no avail.

This might be some sort of interference with OctoDash querying the API or an interaction with any of the plugin dependencies.

**To Reproduce**
1. Install PSUControl and enable the automatic shutdown feature
2. Install OctoDash

**Expected behaviour**
Automatic shutdown after a finished print

**Actual behaviour**
The printer is kept powered on

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - OctoDash Version 1.3.1 and 1.3.2

**Additional context**
Also posted in the forums here: https://community.octoprint.org/t/psucontrol-automation-stopped-working/12906/1

I understand this is vague, so feel free to close this issue if you feel it does not belong here. Unfortunately the PSUControl repo does not allow issues.

UnchartedBull, 2019-10-25:
I think my best guess would be to see if this gets logged somewhere in the system and work my way up from there:
https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/__init__.py#L410

I quickly read over the code and there does seem to be nothing, that is triggered by an API command. OctoDash itself does not send any GCodes to the printer. Maybe try removing all the plugins required by OctoDash and look if it works that way.

hessius, 2019-11-05:
The issue seems to lie within PSUControl and it's management of on off state. It was merely a correlation in time with the OctoDash install, not causation. Closing

closed 2019-11-05 by hessius