UnchartedBull/OctoDash #259: tune temperatures from status screen

mvoss96, 2019-10-25:
I think it would be a great feature to be able to tune the temperatures and the fan from the status screen.
When you touch f.e. the hotend icon it should show a slider where you can set the desired temperature.
Ideally the status screen should also show the current target temperature and not only the current temperature.

UnchartedBull, 2019-10-25:
OctoDash will show the target temperature if there is one. If the target temperature is 0 this will be omitted. You can also adjust the temperatures during print if you just tap the screen and go to the adjust menu.

I may consider adding something for adjusting if you're not printing in the future though ... I'm currently trying to postpone most of the new features so v2 will be ready and can be released finally.

harleyg321, 2020-03-20:
I hadn't actually seen this beforehand but I've just submitted a PR(#503) that would do this.

closed 2020-03-30 by UnchartedBull