UnchartedBull/OctoDash #260: temp not display..

sacy80, 2019-10-26:
Hi ,

i try to install on my octoprint. but it didnt show any temp info.. it always display zero..

UnchartedBull, 2019-10-26:
That‘s weird. Does OctoPrint Display temperatures in the WebUI? Do you maybe have multiple extrudes?

sacy80, 2019-10-26:
yup.. octoprint do show temp of my hotend.. api call issue?? 
i only got 1 hotend :)

sacy80, 2019-10-26:
i try this, and i get return...  tool0
http://10.10.10.209:5000/api/printer/tool?apikey=my api keys
{
  "tool0": {
    "actual": 26.44, 
    "offset": 0, 
    "target": 0.0
  }
}

UnchartedBull, 2019-10-26:
Hmm your API result should be just fine. I got the same results and they work just fine for me. Could you maybe try and open the developer console and see if there are some errors listed there? I think it's Ctrl+Shift+I

sacy80, 2019-10-27:
ok i saw alot of error .. 
[image]
[image]

UnchartedBull, 2019-10-27:
Ok there is a few weird things going on:

Is OctoDash always showing connecting in the bottom right?

Is the other screen always shining through for you or is this just on the image.

Could you send me the whole output of the /api/printer?apikey=...? Then I can try to figure out what is going wrong :)

sacy80, 2019-10-28:
This is the output :)

{
  "sd": {
    "ready": true
  }, 
  "state": {
    "flags": {
      "cancelling": false, 
      "closedOrError": false, 
      "error": false, 
      "finishing": false, 
      "operational": true, 
      "paused": false, 
      "pausing": false, 
      "printing": true, 
      "ready": false, 
      "resuming": false, 
      "sdReady": true
    }, 
    "text": "Printing"
  }, 
  "temperature": {
    "tool0": {
      "actual": 200.0, 
      "offset": 0, 
      "target": 200.0
    }
  }
}

UnchartedBull, 2019-10-28:
What do you mean with that script? It should automatically use the current user in the newest version :).

Ok, I See that you don‘t seem to have a heated bed?

sacy80, 2019-10-29:
something like this.. this should not cause any issue to the script right ? 

if [ -d "/home/octopi/OctoPrint/venv" ]; then
    DIRECTORY="/home/octopi/OctoPrint/venv"
elif [ -d "/home/octopi/oprint" ]; then
    DIRECTORY="/home/octopi/oprint"

UnchartedBull, 2019-10-29:
Nope should work if your user is octopi. The script will ask you for a directory if it doesn‘t find the pi home folder. So both ways will work :)

sacy80, 2019-10-29:
so i wait for next build to be release with the fix for the temp issue ? :)

PR #325 (2019-11-26): Fix issue with no heated bed
Closes #260
closed 2019-11-26 by UnchartedBull