UnchartedBull/OctoDash #262: How do you start Octodash manually from command line?

spiff72, 2019-10-28:
**Describe the bug**
When attempting to start Octodash manually (not launching it at bootup), I get an error stating "Gtk-Warning **: cannot open display"
I tried running ratpoison first, and get "Can't open display"

I don't think this is a bug - just my lack of knowledge on how to run the application.  The pi is already booted, and the console window is currently displayed on the touchscreen.  I suspect that this is the issue (the screen is already running and acting as the console display).

I am running 1.3.2 (to the best of my knowledge), on octopi 0.16.0 and octoprint 1.3.12 on a Raspi3  connected via HDMI and USB (for touch input) to a 7" cap touch display (1024 x 600 resolution).

UnchartedBull, 2019-10-28:
You have to do export DISPLAY=:0 first. After that it should work :)

DAVEfromCZE, 2019-11-10:
Hi, i try run on 3.5inch 480x320 GPIO lcd but i have same error "Gtk-Warning **: cannot open display" i try command " export DISPLAY=:0" but afrer reboot is on screen command line with" -bash: xinit: command not found".I  try octosceen, Octiprint-TFT, and repetier all works, bud your project is much better. Thank you

UnchartedBull, 2019-11-10:
Have you installed OctoDash via the installer script? Please make sure that xinit is installed ...

DAVEfromCZE, 2019-11-13:
Hi yes, my fault :-)
its working now i must install this for (3,5" LCD to GPIO) :

> sudo raspi-config
And navigate to Boot Options - Desktop CLI - and choose Console Auto Login
> git clone  https://github.com/waveshare/LCD-show.git 
> cd LCD-show/
> sudo chmod +x LCD35-show
 > sudo ./LCD35-show
> wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
>sudo apt-get install xinit
> sudo apt install xserver-xorg xinit xserver-xorg-video-fbdev
> sudo systemctl set-default graphical.target mozna neni treba ale zkusit
> sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf /usr/share/X11/xorg.conf.d/99-fbturbo.conf.old

closed 2020-01-20 by stale[bot]