UnchartedBull/OctoDash #266: Needs a simple install for a Noob like me :)

nedla1969, 2019-10-28:
Best looking TFT Ui, how simple is it to install..

UnchartedBull, 2019-10-28:
Just execute this line of code on the pi (either via ssh or via an attached keyboard):

```
wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
```

*This will install additional OctoPrint plugins, listed here: https://github.com/UnchartedBull/OctoDash/wiki/Installation#automatic-installation*

UnchartedBull, 2019-10-28:
This will also install  a new Desktop and configure automatic start on boot. Do you currently have a Desktop running on your system?

UnchartedBull, 2019-10-29:
yes. You may want to uninstall octoprint-tft first to avoid any weird behavior ...

UnchartedBull, 2019-10-29:
If you managed to install OctoDash please close this issue :)

closed 2019-10-29 by nedla1969

evannadeau, 2020-01-21:
I just went through the setup on a fresh 0.17.0 deployment of OctoPi, and the automated version did not work.  I had to configure the console autologin, as well as the very last step in the manual config, sudo chmod ug+s /usr/lib/xorg/Xorg

UnchartedBull, 2020-01-22:
Thanks for the info. Those two steps will be always manual effort. I won't automate the automatic login, because that can be a huge security risk and I want the users do note, that they open up the system to anyone that can plugin a power chord.

The chmod thing only occurs on some installations, therefore it is not in the main script. I'll make sure, that it is more obvious in the wiki which steps needed to be done manually.

evannadeau, 2020-01-22:
Fantastic, thanks.