UnchartedBull/OctoDash #267: Android App can't be installed [Parse Error]

TheAndi, 2019-10-29:
**Describe the bug**
Hey just wanted to get a look and feel of this nice interface and therefore wanted to try the Android APK. Unfortunately it gives me a parse-error while installing "There was a problem parsing the package"

**To Reproduce**
Download APK try to install

**Expected behavior**
App installs

**General Information:**
Hardware: Galaxy S7 running stock Android 8.0.0

Just wanted to let you know the problem. Would also be nice to mention the Android version in the ReadMe to let people know it exists.

Keep up the great work!

UnchartedBull, 2019-10-29:
I don't have any android devices to test this on. This appears to be a common issue with Electron Apps though. Here is what I found:

https://github.com/CITCEuraRFID/KeyDuino/issues/3

Maybe you could try and do the steps mentioned in that issue? Especially the "install apps from unknown sources". If that doesn't work I need to check whether I can revive my old android tablet.

I'll make sure to add this to the README or wiki.

TheAndi, 2019-10-31:
Thanks for your reply. "install apps from unknown sources" is already enabled.
It happens with the 1.3.1 and 1.3.2 release. Let me know if you need further information.

UnchartedBull, 2019-10-31:
Ok I will try to revive my old android tablet and test some stuff to see how this works. Thanks for the info!

UnchartedBull, 2019-11-26:
I did some research and there is zero documentation available on this bug, and about 2 lines of docs how to build an apk with electron-builder. There was an issue opened in their repository, which was closed with no response though ... I'm removing the apks for now and may have a look at this later. Closing this issue for now, as Android support is (currently) dropped. Sorry!

closed 2019-11-26 by UnchartedBull