UnchartedBull/OctoDash #281: Shutdown Button

FlyingT, 2019-10-31:
Nothing fancy, just a Shutdown / Restart Button for the Raspberry Pi

UnchartedBull, 2019-10-31:
https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions. Already possible :)

closed 2019-10-31 by UnchartedBull

FlyingT, 2019-11-01:
Totally missed that, thanks!
I wrote an "popular" [guide ](https://github.com/FlyingT/OctoTouch/wiki/Complete-Setup-Guide) for Prusa Printers and Touchscreens while ago, I think I will scrap Octoprint-TFT and OctoScreen from it and replace it with your OctoDash. Just works like a charm on 3,5" GPIO screens!

UnchartedBull, 2019-11-01:
That would be great! If you need any more information let me know :).

I-like-stuff, 2020-05-24:
Personally I would prefer to have a dedicated shutdown button on the menu page, maybe in the bottom right hand corner.