UnchartedBull/OctoDash #286: Help

nedla1969, 2019-11-03:
Not sure whats happened. 

just restarted and this came up

[image]

UnchartedBull, 2019-11-03:
You can either edit your config file with nano to match this file: https://github.com/UnchartedBull/OctoDash/blob/master/sample.config.json with `nano ~/.config/octodash/config.json`.

 Or just delete it `rm ~/.config/octodash/config.json` and restart your Pi and create a new config.

What did you do prior to this error showing up?

nedla1969, 2019-11-04:
Many thanks,  i was trying to setup my DHT22 sensor, And must of deleted the config file some how..

closed 2019-11-04 by nedla1969