UnchartedBull/OctoDash #291: Advice please

nedla1969, 2019-11-05:
I have a dht22 sensor working perfectly showing the enclosure on octoprint, Is it suppose show on the touch scree to?, Thanks

UnchartedBull, 2019-11-05:
Yes if it is configured. Currently you need to either set this up during startup or manually in the config file. The next update probably will include a way to this directly via the Interface ...

nedla1969, 2019-11-06:
I set it up at start up,  
 `` "octodash": {
                        "touchscreen": true,
                        "temperatureSensor": {
                                "ambient": "dht22",
                                "filament1": null,
                                "filament2": null

UnchartedBull, 2019-11-06:
Is that the name you gave the sensor? You don't enter the type there, but rather the name of the sensor. The name will be displayed in the very left column if you have the enclosure tab open. It works just fine for me ...

nedla1969, 2019-11-06:
Thankyou all sorted

closed 2019-11-06 by nedla1969