UnchartedBull/OctoDash #312: Adding Manual Bed Leveling Screen

TheAndi, 2019-11-18:
Hey, as this project looks so nice and covers lots of features, I am thinking about entirely replacing my MKS Touch Display of my Sidewinder X1.
But there is one feature I am missing right now: 

Bed Leveling like this:

[image]

I know I could add the appropriate five gcode commands to the six custom actions, but this would exclude a lot of other nice commands. 
So a separate Screen (like shown above) for that would be awesome.

Maybe the settings for the spots and speeds can be retrieved from the [Bed Leveling Wizard Plugin,](https://plugins.octoprint.org/plugins/bedlevelingwizard/) so no manual adjusting of the config would be necessary. 

Thanks very much in advance keep up the great work!

UnchartedBull, 2019-11-18:
Good idea, I'll work on this once the v2 release is finished. Currently I don't have much time to work on OctoDash, because the normal day job eats up the majority of my time ... But this is on the list now :)

ThxAndBye, 2020-10-10:
I'd also like to use OctoDash as my only interface on the printer and as such it would be nice to initialize manual bed leveling from OctoDash. Is it still a consideration or maybe already possible?

UnchartedBull, 2020-10-11:
Currently not possible (maybe via Custom Actions, but that's not really ideal I think). Moved this to v3 release, so it should hopefully be worked on soon :)

SaKiEQ, 2021-09-21:
Howdy, 

Any news on this ? 

A potentially simpler version could be, to somehow tap into the Octoprint Bed Level Wizzard, which works great and does what I need most the time, yet pushing that one button to advance to the next level point requires me to have a tablet or laptop next to the printer, next to the OctoDash touch screen.... needless to comment more.

If you could implement an Action Command ([Wizzard-Next-Point]0 that just emulates the action wizards next button command, could be 75% of the problems solution ?

C0untZero, 2022-02-14:
Any news on this feature request? :)

Bond-Addict, 2024-07-02:
Any update on this feature request? I also agree this would be awesome. Being able to use the bed level visualizer in octoprint is nice, but IMO it would be better to be able to view the data points on the actual pi output like you can see on the oem displays for a lot of newer printers.