UnchartedBull/OctoDash #318: Raspberry Pi 2 compatability / Illegal Instruction

Malakai-Mods, 2019-11-21:
Seems the Pi 2 is not compatible with Octodash. Ratpoison runs but stays at a blank black screen. I've reinstalled both by script and manually, updated distro, installed tons of xinit dependencies. I don't know enough linux to debug this further. 

The same sd card image works on a Pi 3

pi@octopi:~ $ octodash
Illegal instruction (core dumped)

uname -a
Linux octopi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

pi@octopi:~ $ lscpu
Architecture:          armv6l
Byte Order:            Little Endian
CPU(s):                1
Socket(s):             1
Model:                 7
Model name:            ARMv6-compatible processor rev 7 (v6l)
CPU max MHz:           700.0000
CPU min MHz:           700.0000
BogoMIPS:              697.95
Flags:                 half thumb fastmult vfp edsp java tls

Core dump doesn't tell me what went wrong. 
[New LWP 1416]
Core was generated by `octodash'.
Program terminated with signal SIGILL, Illegal instruction.
#0  0x0285b020 in ?? ()

Malakai-Mods, 2019-11-21:
Stupid me just realized this isnt a Pi2 its a 1 B+. The armv6l cpu architecture denotes this.

closed 2019-11-21 by Malakai-Mods

UnchartedBull, 2019-11-22:
Yes, Electron can‘t build for armv6l ... I think it will also introduce performance issues I think if you print complicated G-Code as the Pi1 even struggles with plain OctoPrint, at least for me ...