UnchartedBull/OctoDash #354: Touch UI x OctoDash

Testmangh, 2019-12-23:
**Describe the bug**
I installed the Octodash but nothing has changed... I still have the Touch UI screen on the LCD.
Do I need to uninstall the touch UI plugin?

**To Reproduce**
With Touch UI + Touch screen working normally just install the Octodash.

**Expected behavior**
Octodash interface on the touch screen.

**Screenshots**
https://imgur.com/a/A9s1mIS

**General Information:**
 - Hardware Raspberry Pi Model 3 B +
 - OctoPrint version: 1.3.12
   OctoPi version: 0.16.0
 - OctoDash Version: 1.3.3

**Additional context**
I'm noob, sorry if this is not the place for the question...

UnchartedBull, 2019-12-27:
I never tried the Touch UI, could you maybe try uninstalling the Touch UI? I don‘t know exactly how Touch UI initialized ...

Testmangh, 2019-12-27:
Ok I will do that... It will take a while because I burned all my drivers and the SKR 1.3 board xD... Thanks for the answer...

closed 2020-03-29 by stale[bot]