UnchartedBull/OctoDash #368: Printer Reset During Print after touching black screen

spiff72, 2020-01-12:
This may have been an odd situation, but I had an unfortunate error occur today.

I had turned off the printer (Prusa Mk3s) to make some adjustments, and turned it back on.  I connected back to the printer in my Octoprint server browser tab, and started a print.  

About 30 minutes into the print (24 hours long), I noticed the Octodash screen was black, so I touched it to wake it up.  When I did this, it immediately rebooted the printer and I lost the print I had going.  I am VERY glad it was only 30 minutes in and not 23:30 in!  I am not sure if it was sleeping or what state it would have been in at this point, but I am making a mental note to NEVER touch a blank screen to wake it up if I have a print running, since it never goes blank when a print is running (or even when a print is not running and a the Octodash application is active).

Is this normal behavior or did I find an issue here?

UnchartedBull, 2020-01-12:
Ok, that is weird ... OctoDash only allows restarting the Pi via the Custom Actions, there is no other functionality available, that would enable OctoDash to restart the Pi ...

The screen can be turned off if the paramter "turnScreenOffSleep" is set to true. This is false by default and not documented anywhere so far. Maybe you could check your config to see, whether this option is set to true?

I have a look at this, but I've never experienced any issues similar to this ...

spiff72, 2020-01-12:
I will check for that setting. I should also add that it didn't reboot the pi. It appeared to reset or power cycle my printer. It acted like it disconnected from octopi and then reconnected.

spiff72, 2020-01-12:
I just checked, and I DO have the turnScreenOffSleep option set to true.

UnchartedBull, 2020-01-13:
Ok ... This means, that OctoDash will turn off the screen, once its sleeping. Although it shouldn't go sleeping, unless the printer is disconnected ... If OctoDash was sleeping once, it waits until you tap the screen to wake everything back up. This will also send a connect message to OctoPrint. In my testing connecting to the printer, while already being connected didn't have an effect. I may have to recheck that. What OctoPrint Version are you running?

spiff72, 2020-01-13:
Octoprint 1.3.12 running on OctoPi 0.16.0

UnchartedBull, 2020-01-14:
Ok I'll have a look, whether I can reproduce that.

spiff72, 2020-01-14:
OK - thanks.  There have been a few other strange behaviors lately with OctoDash too.  The font seems slightly different than I remember, and I have had a few occasions where I run a print, and it says "No Job Running" in the top section even though there is a print job running.  This these cases, the temperature readings were correct on the screen and the status in the lower corner would state "Printing".  I haven't updated it in quite some time, so I don't know why these would happen.

UnchartedBull, 2020-01-14:
I did change the font of the heading, the rest should be the same. If your Pi doesn't have an internet connection the fonts don't get downloaded though ...

spiff72, 2020-01-14:
It does have an internet connection, but would the fonts be updated without running the update script?

UnchartedBull, 2020-01-22:
Hi, I just confirmed your issue. The OctoPrint API states, that the connect command will trigger a reconnect if the printer is already connected. http://docs.octoprint.org/en/master/api/connection.html#issue-a-connection-command. Don't really know why OctoDash went into sleeping mode in the first place though ...

I'll make sure to include a test, whether OctoPrint is already connected in the next update. Sorry for you cancelled print!

spiff72, 2020-01-22:
> Hi, I just confirmed your issue. The OctoPrint API states, that the connect command will trigger a reconnect if the printer is already connected. [...]

No problem - it was actually lucky timing since I did this shortly after the print started!

jneilliii, 2020-03-04:
@UnchartedBull, has this change made it into the latest release version of OctoDash?  I ask because this just happened to me.  Luckily I wasn't in the middle of a print, but wanted to use the OctoDash to move the head around, and my printer reset.

UnchartedBull, 2020-03-13:
it did, but another user reported a similar issue. Maybe something with the state checking isn't working as expected, leading OctoDash to trigger a reconnect command anyway. Will have a look at this.

jneilliii, 2020-03-20:
@UnchartedBull, I see you've merged a fix for this issue and I was wondering if you'd like to get some beta testers? If so, just let me know the process to build the deb package, or instructions for manually updating from source I'd be happy to help.

closed 2020-03-31 by UnchartedBull

spiff72, 2020-03-31:
Is the best way to update the script described in the readme (it will always grab the newest release)?

UnchartedBull, 2020-03-31:
Yes script is the preferred way. It will always grab the latest release available :)

UnchartedBull, 2020-03-31:
@jneilliii sorry I just saw your comment. Feel free to test around, but it should be fixed now (hopefully). If you want to build OctoDash from source just run `npm run pack` which should take care of everything and provide the deb files in the package folder. You have to run the obligatory `npm install` before though :)

PR #378 (2020-01-22): Fix/standby improvements
Improve standby, still not perfect, but killing the singletons is currently out of scope for this project. Notifications are catched earlier now and the console is cleared on a reconnect, which should clean up the possible garbage.

Connection is checked before triggering a connect command to prevent restarting. Notification are muted until the connection is established, which should prevent random errors popping up.

PSUControl is also added natively now.

Closes #214 
Closes #368
Closes #371
PR #531 (2020-03-31): fix/standby-reconnecting
Fixes #368.