UnchartedBull/OctoDash #371: Constantly throwing connection errors.

mikekscholz, 2020-01-13:
**Describe the bug**
I use PSU control and switch my printers mainboard through octoprint. I have a custom version of portlister installed with 0 second lag time so the instant marlin receives power octoprint starts connecting, which immediately causes marlin to reboot which is normal behavior. I modified the typescript for the standby module so that when you press the screen to wake, it sends the PSU on command which is working perfectly, but its throwing error messages before its even had 3 seconds for marlin to finish booting. Also as an aside, if I open octoprint from my laptop and click to power on the power supply, I still get a 409 error that pops up on top of the standby screen. then once its properly connected and the main screen is properly showing live data I usually get 2 more error popups that I need to clear before they stop. Im kind of at my limit for typescript syntax and free time is scarce for learning a new language at the moment, but is there any way to get octodash to chill out for like 10 seconds when a connect command is issued?

Here is the relevant section of code of my standby mods:
```typescript
import { Component, OnInit } from '@angular/core';
import { ConfigService } from '../config/config.service';
import { Subscription } from 'rxjs';
import { HttpClient, HttpErrorResponse } from '@angular/common/http';
import { Router } from '@angular/router';
import { AppService } from '../app.service';
import { PsuControlService } from '../plugin-service/psu-control.service';

@Component({
  selector: 'app-standby',
  templateUrl: './standby.component.html',
  styleUrls: ['./standby.component.scss']
})
export class StandbyComponent implements OnInit {

  connecting = false;
  error = '';
  httpPOSTRequest: Subscription;

  constructor(
    private configService: ConfigService,
    private http: HttpClient,
    private router: Router,
    private service: AppService,
    private psuControlService: PsuControlService) { }

  ngOnInit() {
    if (this.configService.getAutomaticScreenSleep()) {
      setTimeout(this.service.turnDisplayOff.bind(this.service), 300000);
    }
  }

  reconnect() {
    this.connecting = true;
    this.psuControlService.changePSUState(true);
    if (this.httpPOSTRequest) {
      this.httpPOSTRequest.unsubscribe();
    }
```
**General Information:**
 - Hardware: Pi 3b+ connected to 1024x600 hdmi capacitive touch screen
 - OS Info: OctoPrint 1.3.12 running on OctoPi 0.17.0
 - OctoDash: selfbuilt from master branch with and without the above mods as well as 1.3.3 official release.

UnchartedBull, 2020-01-14:
This is related to #214. Standby currently just omits all errors, but once you tap the screen they will be shown again. Ideal would be, that everything just shuts down until you restart. I'll have a look, whether there is a quick and dirty way to handle this, until I have time to do it properly.

UnchartedBull, 2020-01-22:
I've included the PSUControl in the main code. With the next version you should be able to specify this via the config. Also the notifications are now suppressed for the first 4 seconds after connection, which should hopefully fix the spam :) Still needs some testing though ...

mikekscholz, 2020-01-22:
I figured adding a config parameter to enable this would be the way to go but I had only gotten about halfway to figuring out how to go about it after learning how to include it properly in the first place. Ill do a pull and rebuild and give her a go once I get out of the office for the day.

UnchartedBull, 2020-01-22:
The stuff with the errors being thrown during connection is fixed now. The config parameter will currently only return false until the update with the new config is merged.  So you could either change that line in the config service or wait until the release and stay on your build for now . :)

closed 2020-01-22 by UnchartedBull

mikekscholz, 2020-01-24:
This did the trick :) Did a build with the config param changed to return true and everything works as I would expect now.

PR #378 (2020-01-22): Fix/standby improvements
Improve standby, still not perfect, but killing the singletons is currently out of scope for this project. Notifications are catched earlier now and the console is cleared on a reconnect, which should clean up the possible garbage.

Connection is checked before triggering a connect command to prevent restarting. Notification are muted until the connection is established, which should prevent random errors popping up.

PSUControl is also added natively now.

Closes #214 
Closes #368
Closes #371