UnchartedBull/OctoDash #379: Will OctoDash work with USB based touch screens?

evannadeau, 2020-01-22:
I'm looking at getting this screen...

https://www.waveshare.com/5inch-hdmi-lcd-h.htm

and it appears to connect to the pi for touch via USB.  I believe each touch would compute to a single mouse click, so I believe this will work, but I thought I would reach out first to ask if you have used a screen of this type before.

evannadeau, 2020-01-22:
One more thing, this is a single touch screen.  Will that be an issue for any current or upcoming OctoDash features?

UnchartedBull, 2020-01-22:
I think one guy uses the waveshare screen as well and it works for him. I don't have a screen like this so I can't 100% confirm this, but it should work.

Single Touch Screen isn't an issue. There are currently no features planned that rely on multiple inputs. Currently only Clicking, Scrolling and Holding is needed, which is supported by most touchscreens.

UnchartedBull, 2020-01-22:
https://www.prusaprinters.org/prints/18078-octo-prusa-mk3s maybe you can check out this post, he also uses a waveshare touchscreen

iz2k, 2020-01-22:
Hello, I am the publisher of the octo-prusa-mk3s post. For my setup I used a 3.5'' SPI LCD from waveshare. I only had to install the drivers provided by the manufacturer and everything worked fine.

In your case, the selected display uses the HDMI interface for the display and USB for touchscreen, so it should be even easier to make it work. The display will work out of the box, and probably also the touchscreen, as it is just an HID input device. You should be fine to run octodash on it. The only drawback is that you will need to wire up one hdmi cable and another USB cable. You could also try one of the DSI interface versions to reduce the wiring: https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-2/4.3inch-dsi-lcd.htm

Aditionally, I would recommend to install unclutter to hide the cursor. It works great on raspian.

evannadeau, 2020-01-22:
Thank you both for the replies.  Yeah, I think it will be super easy to use.  What I do see to be an issue is the cables coming out from the side of the display.  I am looking into adapters and cables with 90 degree bends on the end, so hopefully, I can clean up the cable routing.

closed 2020-01-22 by evannadeau