UnchartedBull/OctoDash #385: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/n/nss/libnss3_3.42.1-1+deb10u1_armhf.deb

evannadeau, 2020-01-25:
I am performing a fresh manual install on a fresh Octoprint 0.17.0 while trying to figure out my LCD display.  I receive the error menionted in the heading,  Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/n/nss/libnss3_3.42.1-1+deb10u1_armhf.deb

I checked that source and what is available is libnss3_3.42.1-1+deb10u2_armhf.deb  (note u2 instead of u1).  the package libnss3_3.42.1-1+deb10u1_armhf.deb is no longer available.

evannadeau, 2020-01-25:
OK, maybe I found the solution.  It is the install of apt install xserver-xorg that is missing this I believe.

`pi@octopi:~ $ sudo apt install xserver-xorg
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 octodash : Depends: libgtk-3-0 but it is not going to be installed
            Depends: libnotify4 but it is not going to be installed
            Depends: libnss3 but it is not going to be installed
            Depends: xdg-utils but it is not going to be installed
            Depends: libatspi2.0-0 but it is not going to be installed
            Depends: libappindicator3-1 but it is not going to be installed
            Depends: libsecret-1-0 but it is not going to be installed
 xserver-xorg : Depends: xserver-xorg-core (>= 2:1.17.2-2) but it is not going to be installed
                Depends: xserver-xorg-video-all but it is not going to be installed or
                         xorg-driver-video
                Depends: xserver-xorg-input-all but it is not going to be installed or
                         xorg-driver-input
                Depends: x11-xkb-utils but it is not going to be installed
                Recommends: xserver-xorg-legacy but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
pi@octopi:~ $ sudo apt --fix-broken install
`

I did the apt --fix-broken install, but that installed a lot of stuff.  thinking back, I think this may have been what I did on my first install.  However, perhaps it would be better to use the ` Recommends: xserver-xorg-legacy but it is not going to be installed
`
Continuing to diagnose.

evannadeau, 2020-01-25:
OK, that has resolved it.
This step fails.

`sudo apt install xserver-xorg --no-install-recommends ratpoison x11-xserver-utils xinit`

performed `sudo apt --fix-broken install`
This installed a lot of stuff.

Then re-ran `sudo apt install xserver-xorg --no-install-recommends ratpoison x11-xserver-utils xinit`
It installed a bunch more.

Finished it off with correcting console 2, permission denied message
`sudo chmod ug+s /usr/lib/xorg/Xorg`

Console is now running again, although I don't have my screen touch working (my issues).  I'll leave missing pieces of the install to you.  Please let me know if you want me to test anything.

UnchartedBull, 2020-01-26:
Yeah I forgot to update the apt get install command for the manual installation. Main issue for you was the missing libgtk-3-0. I'll update the wiki, thanks for the info!

Is your OctoDash now running?

evannadeau, 2020-01-26:
Yep, I'm running fine. Just waiting on my screen to arrive. Thanks for the response.  Just wanted to bring it to your attention to help resolve issues for the community at large.  I'll leave the ticket open for you to close when you feel it is resolved on scripts and documentation. I'm happy to help in any way necessary, including with documentation. Thanks for the great work.

UnchartedBull, 2020-01-29:
I'll make sure to update and improve the docs once I'm closer to release v2. Currently I don't have much spare time for the project, so if you want to improve the docs feel free to create a PR for that, would gladly appreciate that!

evannadeau, 2020-01-29:
I will do what i can. I think this is the best looking dashboard for Octoprint, so I want to see it go forward.

UnchartedBull, 2020-01-31:
Yeah I also want to finish that project, but currently there is not much time to spare. I'll update the issue name and type to make it easier to track :)

PR #705 (2020-06-02): Update Installation Script
- [x] update wiki

Closes #385
closed 2020-06-02 by UnchartedBull