UnchartedBull/OctoDash #386: Option to disable sleep screen

kvandt, 2020-01-26:
My tft does not have a touch feature, so when OctoDash goes into sleep mode, I cannot wake it up. I use OctoDash only for status display directly at my printer.

It would be nice to have to option to disable the sleep screen in config. I do not mind to have the status screen on all time.

UnchartedBull, 2020-01-29:
I'll have a look at that.

UnchartedBull, 2020-02-05:
Maybe to clarify a few things:

You turn off your printer and then OctoDash would go into sleep mode, right? Would showing the sleep screen and automatically exiting that once a connection is established (could take around 10s to wake up, based on the polling interval)? The problem is, that the Status Screen would not really work, because there is no data available, that could be shown.

kvandt, 2020-02-05:
My usecase is very simple. If I power up my printer, the raspberry is powered as well (same supply) and starts Octoprint. When I finished printing I turn off my printer and Octoprint (all by a smartplug). So I prefer to see the status of my printer on octodash and since I do not have a touch screen, I cannot get out of sleep modus. My controller board is booted way quicker than my raspberry, so it should always be able to connect and show status.

UnchartedBull, 2020-02-05:
I have a similar setup to yours (except the touchscreen) and I never see the Sleep Screen. Does your OctoPrint automatically connect to the printer? Or do you need to press connect each time?

kvandt, 2020-02-06:
It connects automatically, buy i still get sleep screen. When it starts printing it comes out of sleep. I would just prefer to disable the sleep screen...

UnchartedBull, 2020-02-10:
If the printer connects automatically it never should go into the sleep screen in the first place, I'll have a look why this could happen ... Would you mind sharing some more information about your setup, especially version?

kvandt, 2020-02-10:
Strange, since this weekend i don't see the sleeping screen anymore. It now starts with "no job running". I used to have several errors on connecting to the printer (and therefore sleep screen). I did not change anything on the setup (running Octoprint 1.3.12, Octodash 1.3.3) except for updating PrintTimeGenius last Firday. Do not understand <confused>. So it seems it works as it should. When printer is connected it shows "no job running". Funny thing is that I printed something yesterday through API from PrusaSlicer, where printer starts but Octodash still reports "no job running". I think this issue can be closed (even though I do not know what changed).

UnchartedBull, 2020-02-11:
Hm, funny that it now works. I'll close this issue and keep track of the no job running in issue #411. If this error occurs again please reopen this issue :)

closed 2020-02-11 by UnchartedBull