UnchartedBull/OctoDash #388: Working great, but some interface suggestions

evannadeau, 2020-01-27:
OctoDash version 1.3.3
OctoPrint 1.3.12
OctoPi 0.17.0

**FILES**
Once clicking on a file in the files list, my only two options are to press the thermometer icon to start pre-heat, or to press the print button.  There is no way to get out of the selection of that file, without pressing print, and then right away cancelling.

What I do really like is the press and hold option that comes up, I feel this should actually be the single click default screen.  It gives you the option to delete it right there, or to press the button to print.  It ends up being the same number of clicks to start a print job.

**CONTROL**
The Heatup and Cooldown buttons, I feel should take you back to the home screen.  Or at least give some feedback that something is occurring.  If it was on the home screen, you would see the temps changing, but sitting on the control screen, you don't know anything is happening.  However, perhaps all the buttons here should give feedback that something is occurring, perhaps with a pop-up.  Especially for the reboot and shutdown buttons.  Right now they take effect without any confirmation prompt and immediately reboot or shutdown your Octoprint.

UnchartedBull, 2020-01-29:
You can close a file with swiping from right to left on that file, not that obvious I know. I get your point with the same number of clicks to start a print, I did the other action based on feedback from another guy. Now that there is feedback for each version of this I may revisit it. Thanks for the feedback :)

Control is related to #225 and will be tracked there :)

evannadeau, 2020-01-29:
I'll give that a try to close the file. Thanks.

When I can get some free time, I'll see about making some changes myself and doing a pull request.

StealthDefender1, 2020-01-29:
press and hold don't work for me?! Also i get a error about layer plugin or something. Like the look from the octodash. nice work.

UnchartedBull, 2020-01-29:
Press and Hold only works in the file list (not in the screen where the file is loaded). In the loaded file screen you need to swipe from right to left to remove the file and get the menu back.

Layer Plugin Error is related to an issue in the DisplayLayer Progress Plugin and is tracked in OllisGit Repository. Will be resolved with the next release of his Plugin, can't do anything about that, sorry

StealthDefender1, 2020-01-29:
@UnchartedBull thx for infos. 
i mean in the file list. When i press and hold a file nothing pops up. Don't know why? maybe with a next release. A nice feature could be to see the temp graph like in touchui. The rest is amazing and very usefull.

evannadeau, 2020-01-29:
@StealthDefender1 what version of OctoDash are you running?

StealthDefender1, 2020-01-29:
1.3.3, the latest i think.

evannadeau, 2020-01-29:
And can you share what type of screen you have, please?  Perhaps the screen is not sending a proper hold command to OctoDash.

StealthDefender1, 2020-01-29:
@evannadeau this one
STARTO Raspberry Pi Touchscreen 3,5 Inch Zoll Display TFT Monitor LCD 320x480 Auflösung mit Gehäuse Case und 3 Stück Heatsink Kupferkühlkörper, Touch Pen für Raspberry Pi 3B + / 3B / 2B + / 2B TS11 https://www.amazon.de/dp/B07S8CKW58/ref=cm_sw_r_cp_apa_i_C7EmEbJY4T8NN

UnchartedBull, 2020-01-31:
Those displays that are connected via the Pi GPIOs are always a bit funky, so it's most likely your display that isn't correctly sending the hold event.

You can test this by holding on one of the files, if the screen changes to the Home Screen BEFORE you release your finger your screen doesn't support hold, if you move to the Home Screen AFTER you release your finger (leave it on for 4-5 seconds) then this is a bug with OctoDash.

UnchartedBull, 2020-02-12:
The file view has been switched around (hold currently is not working anyways for some weird reason). You can also configure automatic switch back to home screen and confirmation dialogs for the custom actions now. This can either be done in  the new settings menu or via the config, once the new update is released.

PR #422 (2020-02-12): Custom Actions Closing and Confirmation working 
Closes #419 
Closes #388
closed 2020-02-12 by UnchartedBull