UnchartedBull/OctoDash #391: Add enclosure temp sensor after initial setup?

spiff72, 2020-01-30:
I recently added a temperature sensor to my enclosure and I think Octodash supports the use of additional temperature sensor(s) (possibly via the enclosure plugin?)?

How can I add this to Octodash?  I am using an 18B20 temp sensor and plan to use PWM speed control on a 12 volt 4-wire exhaust fan on the enclosure (once I get the fan installed and wired up).

tiwo85, 2020-01-30:
You must edit the configfile.
`nano ~/.config/octodash/config.json`
This is my part of the configfile:
`                       "temperatureSensor": {
                                "ambient": 1,
                                "filament1": null,
                                "filament2": null
                        },`

UnchartedBull, 2020-01-31:
@tiwo85 is correct, you need to enter the ID of the sensor from the EnclosurePlugin (most likely 1, if this is your only sensor) into the temperatureSensor.ambient attribute. This will be doable via the UI Settings shortly though ...

closed 2020-01-31 by UnchartedBull

spiff72, 2020-01-31:
Thanks, I will try adding that later today.

Does it matter where in the config file that statement is placed?

UnchartedBull, 2020-01-31:
No need to add that. It is already there just change the ‚null‘ of temperatureSensor to your sensorID.

spiff72, 2020-01-31:
Ah - OK.  Thanks!  I poked around in that file before I posted this question, and didn't remember seeing it, but I could have easily missed it.  I can't get to the Pi until tonight to try it out...