UnchartedBull/OctoDash #397: Show degrees F for enclosure sensor as an option (if it is configured as such in Enclosure Plugin)

spiff72, 2020-02-01:
**Is your feature request related to a problem? Please describe.**
Minor issue - I am using a temperature sensor in my enclosure and just added it to Octodash.  I have it configured for degrees F currently (but may switch it to degrees C in the future for consistency with the extruder and bed temperatures).  I have this configured in the Enclosure Plugin.

**Describe the solution you'd like**
I don't know if there is a way to see the configuration of the enclosure plugin temp sensor, but if not, I suspect it could be assumed if the temp is over say 50 degrees, it can probably be assumed that it is degrees F.

This is relatively minor - and I will likely try to switch to degrees C in the future - I just need to figure out what temps to set up in the Enclosure Plugin for the PWM link to temperature.

UnchartedBull, 2020-02-03:
That's possible. Hopefully there will be some time to spare this weekend to continue work on OctoDash. Leaving this link here for later: https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/API.md#list-a-specific-input

spiff72, 2020-02-03:
As I mentioned above, this should be low on the list of priorities!  I already switched my enclosure plugin to deg C and I am working on some other stuff that will likely require it stays as such.  (Using GCODE to set target temp for a temperature sensor control that toggles an exhaust fan).  I just need to force myself to get used to deg C!  I printed a cheatsheet chart so that should help in that regard. :-)

UnchartedBull, 2020-02-04:
Here in Germany degree Celsius is the normal unit, so everyone is super confused with Fahrenheit. But it‘s an easy fix and should‘ve been there in the first place, thanks for raising the issue!

spiff72, 2020-02-04:
Understood.  Someday maybe the US will adopt the metric system too.  (And will stop being one of two(?) countries who insist on still using English units)...

PR #407 (2020-02-07): Use correct temperature unit
Closes #397
closed 2020-02-07 by UnchartedBull