UnchartedBull/OctoDash #398: Custom icons in "custom actions"

Protoncek, 2020-02-02:
I just discovered this plugin. Currently i use Octoscreen, but i find this one really... beautiful, simple, it contains only what i really need, so i might just switch to this one. 
I understand that it's still under development, but i'd like to know (i searched but didn't find) this: i would like for custom actions to have preheat for PLA, PETG and ABS - i know how to do it, but if i understand correctly it's only possible to use icons from "fontawesome.com" ? Is it possible to use my own? Since that web site doesn't have suitable icons for this purpose.

Thanks!

UnchartedBull, 2020-02-03:
Hi, unfortunately only fontawesome is available right now. Introducing custom icons will require some major code changes, because fontAwesome offers an easy integration with Angular. I'll be happy to accept a PR for this, but I currently have no time to work on this. Will leave this here, so I can pick up on this later maybe.

Protoncek, 2020-02-03:
OK, thanks for explanation. For now i just put icons of thermometer (quarter, half and full) for three filaments and i'll be fine. I'll explore some more on those icons if i find more suitable ones...
Looking forward for future improvements...

JoaoFaelMatos, 2020-03-12:
> Hi, unfortunately only fontawesome is available right now. Introducing custom icons will require some major code changes, because fontAwesome offers an easy integration with Angular. I'll be happy [...]

Tho, is it possible if one would create a .ttf font file with custom logos in it? I know it's not a practical solution but for the moment, it's something I could work on to customise my UI

UnchartedBull, 2020-03-13:
Probably not. I'm using the fortawesome angular integration, which makes it super easy to integrate these icons, but I think the library doesn't accept a ttf font or similar. Once v2 is released I will add support for custom icons though ...

AndersWinther-Dahl, 2022-09-17:
Just a suggestion, how about just letting people paste svg / img elements with source code in them?