UnchartedBull/OctoDash #400: Problem with "DisplayLayerProgress" plugin

Protoncek, 2020-02-04:
I found another "problem": i have Samba installed on my RPi, so i have mapped a network drive in WIN explorer and i upload my g codes from PrusaSlicer directly to a network drive (into RPi), not through drag&drop feature, since it's waaaay easier. And there lies a problem... i get a corrupted file message in OctoDash.  I changed my slicer settings so it already inserts lines:
M117 INDICATOR-Layer0
;Z = 0.2
and i changed height settings in plugin to "Max Z-Value with Extrusion on Z-Layer", so in Octoprint i get correct numbers for layers and height

But still i get error in OctoDash that file is corrupted, HOWEVER only until i reboot RPi. After reboot same file is suddenly ok...? On octoprint web page number of layers and height is immediately correct.
Is there anything to be done in this way? File itself is definitely ok, because, as i said, it's ok in octoprint web page, and it's only "corrupted" if i don't reboot Pi after uploading.

UnchartedBull, 2020-02-05:
Could you maybe post a message of the error that is showing up? I don't know if I can do something about that, because OctoDash does not do any of the computations, but rather only uses the APIs provided by the different Plugins. But I see what I can do

Protoncek, 2020-02-05:
Sure. Here's error message. Since you said that OctoDash does nothing i tried to restart octoprint and  after restart error is gone, so it seems that really octoprint "does something"... 
I attached two files: one BEFORE restart and one AFTER restart of octoprint. I can't find any difference, maybe you're smater here...
Thanks!

[image]

[g-codes.zip](https://github.com/UnchartedBull/OctoDash/files/4161034/g-codes.zip)

UnchartedBull, 2020-02-05:
G-Codes are identical down to the byte ... There is a similar issue that is occurring from time to time for myself. I upload a file via the OctoPrint WebUI and then it doesn't show up in the FileBrowser on OctoDash. Only happens every 20th time or so, that's why I didn't open an issue in the OctoPrint Repo because the error can't be reproduced easily and I didn't had the logs handy.

Would you be able to support an issue I'll create in the OctoPrint Repository with Logs and additional information maybe? If so, I can create the issue later in the evening and hopefully get this fixed.

From OctoDash side there is nothing I can do here really, the API simply isn't returning any information about the file (except that it exists), that's why you're seeing this error ... If you're ok with it, I would close this issue and take this over to OctoPrint.

Protoncek, 2020-02-05:
Sure, i will support It, if it will help to solve this issue. Just let me know what to do an i hope i will know how to do it...

Protoncek, 2020-02-15:
Hello again!
New info: i dug and dug... and found out that octoprint has a watched folder by default, that's "/home/pi/.octoprint/watched" (i found it in octoprint settings "folders"). If i save my g-code in THAT folder it's automatically copied into correct folder by octoprint program (which is "uplads") . And i no longer get an error!!! So, it's necesarry to setup samba that points to a folder "watched".
Then octoprint moves file to correct location. I guess this folder is the one when you drag&drop to web page...
So, basically i kinda solved my problem, only i think it might me good that you mention this on your page for other users.

I hope that i wrote so that you understand (english is not my language), otherwise feel free to ask...
Thanks!
Paul

UnchartedBull, 2020-02-16:
That's great info! Thanks!

I'm closing this issue here, since the Caching Issue with the File API is already tracked in the OctoPrint Repository and there is nothing really I can do here. Thanks for the headsup for users who might run into the same issue :)

closed 2020-02-16 by UnchartedBull