UnchartedBull/OctoDash #409: Where is the config.json actually stored and read?

MatthewCroughan, 2020-02-09:
I'm trying to implement this project into https://github.com/MatthewCroughan/octobalena

However, I cannot understand how the config comes to be generated. I've ran this project on my local machine like so.
`git clone https://github.com/UnchartedBull/OctoDash.git`
`cd OctoDash`
`npm install && npm run ng:build`
`electron .`

This then brings up Octodash telling me I'm running it for the first time.

`~/.config/octodash/` exists after the first run.

Closing the electron instance for OctoDash and providing a valid config file `~/.config/octodash/config.json` does not change this behaviour. No matter what, OctoDash will always run as if it is its first time, despite providing this config.

If I go through the first-run configuration manually and run `find ~/ -name 'config.json` there will be absolutely no evidence of a config.json existing anywhere. Is it possible we could document this together?

MatthewCroughan, 2020-02-09:
Oh, I see! I was providing a modified version of the `sample.config.json` provided by the repo. It didn't like what I had provided. It then appends onto the existing config.json file the correct information, whilst keeping the bad configuration present in the file.

## BEFORE (The config I provided):
```json
{
    "octoprint": {
        "url": "http://192.168.1.210:80/api",
        "accessToken": "D00D3250BEBA4285AD39C990B1967943",
        "apiInterval": 15000
    },
    "printer": {
        "name": "Matt's Ender 3",
        "xySpeed": 100,
        "zSpeed": 5
    },
    "filament": {
        "thickness": 0,
        "density": 0
    },
    "octodash": {
        "touchscreen": true,
        "temperatureSensor": null,
        "customActions": [{
                "icon": "home",
                "command": "G28",
                "color": "#dcdde1"
            },
            {
                "icon": "ruler-vertical",
                "command": "G29",
                "color": "#44bd32"
            },
            {
                "icon": "fire-alt",
                "command": "M140 S50; M104 S185",
                "color": "#e1b12c"
            },
            {
                "icon": "snowflake",
                "command": "M140 S0; M104 S0",
                "color": "#0097e6"
            },
            {
                "icon": "redo-alt",
                "command": "[!RELOAD]",
                "color": "#7f8fa6"
            },
            {
                "icon": "skull",
                "command": "[!KILL]",
                "color": "#e84118"
            }
        ]
    }
}
```

## AFTER (What it does to the provided file):
```
{
	"octoprint": {
		"url": "http://192.168.1.210:80/api",
		"accessToken": "D00D3250BEBA4285AD39C990B1967943",
		"apiInterval": 15000
	},
	"printer": {
		"name": "Matt's Ender 3",
		"xySpeed": 100,
		"zSpeed": 5
	},
	"filament": {
		"thickness": 0,
		"density": 0
	},
	"octodash": {
		"touchscreen": true,
		"temperatureSensor": null,
		"customActions": [
			{
				"icon": "home",
				"command": "G28",
				"color": "#dcdde1"
			},
			{
				"icon": "ruler-vertical",
				"command": "G29",
				"color": "#44bd32"
			},
			{
				"icon": "fire-alt",
				"command": "M140 S50; M104 S185",
				"color": "#e1b12c"
			},
			{
				"icon": "snowflake",
				"command": "M140 S0; M104 S0",
				"color": "#0097e6"
			},
			{
				"icon": "redo-alt",
				"command": "[!RELOAD]",
				"color": "#7f8fa6"
			},
			{
				"icon": "skull",
				"command": "[!KILL]",
				"color": "#e84118"
			}
		]
	},
	"config": {
		"octoprint": {
			"url": "http://192.168.1.210:80/api/",
			"accessToken": "D00D3250BEBA4285AD39C990B1967943",
			"apiInterval": 1500
		},
		"printer": {
			"name": "Matt's Ender3",
			"xySpeed": 150,
			"zSpeed": 5
		},
		"filament": {
			"thickness": 1.75,
			"density": 1.25
		},
		"octodash": {
			"touchscreen": true,
			"temperatureSensor": {
				"ambient": null,
				"filament1": null,
				"filament2": null
			},
			"customActions": [
				{
					"icon": "home",
					"command": "G28",
					"color": "#dcdde1"
				},
				{
					"icon": "ruler-vertical",
					"command": "G29",
					"color": "#44bd32"
				},
				{
					"icon": "fire-alt",
					"command": "M140 S50; M104 S185",
					"color": "#e1b12c"
				},
				{
					"icon": "snowflake",
					"command": "M140 S0; M104 S0",
					"color": "#0097e6"
				},
				{
					"icon": "redo-alt",
					"command": "[!RELOAD]",
					"color": "#7f8fa6"
				},
				{
					"icon": "skull",
					"command": "[!KILL]",
					"color": "#e84118"
				}
			],
			"turnScreenOffSleep": false
		}
	}
}
```

closed 2020-02-10 by MatthewCroughan

UnchartedBull, 2020-02-10:
Yeah that is an electron thing I can do nothing really about. You always have to wrap everything in the "config" attribute. I'll make sure to mention this in the docs for users, who want to create a config manually.

There will be a significant config change in the next release, just as a heads-up :) You can already view the new format here: https://github.com/UnchartedBull/OctoDash/blob/feature/settings-2/sample.config.json (needs to be wrapped in config again)

MatthewCroughan, 2020-02-10:
@UnchartedBull https://twitter.com/MatthewCroughan/status/1226835365413126144

This tweet gets the point across as to what I'm trying to achieve. Your app works as is right now and you can see the commits used to achieve this. I'll update accordingly, but if you have a more realtime mode of contact I'd love to pick your brains about electronisms and nodeisms so I can package this more effecively. Matrix or Riot perhaps? Feel free to PM me on Twitter.

UnchartedBull, 2020-02-10:
Damn that edit is pretty sweet. I might need to hire you for one of OctoDash as well haha :D

I'll write you a PM on twitter so you can get some insight scoops before each release :) We can also talk about packaging once v2 is out of the window. The daytime job is currently eating up the majority of my time, but I'll try to get this done soon. After that we can think of a custom packaging for Octobalena :)

UnchartedBull, 2020-02-10:
I can't seem to work out twitter dm's. I liked your post so it would be great if you could just dm me @timongaebelein, can't seem to find that stupid dm button.

MatthewCroughan, 2020-02-10:
@UnchartedBull Both participants have to follow eachother on Twitter in order to DM. I've followed you, for it to appear you have to follow me.