UnchartedBull/OctoDash #410: Control Screen Backlight Brightness

DeviousPenguin, 2020-02-10:
I'd like to be able to control the brightness of my RPi screen from within Octodash. Currently I'm changing the brightness using SSH:

```sh
echo 127 | sudo tee /sys/class/backlight/rpi_backlight/brightness
```

to set screen to half brightness, for example. Feel free to try this out if you have the official RPi display.

A simple way to do this would be to allow the Custom Action buttons to run shell commands, but a simple brightness slider (similar to a mobile phone) would be far nicer. Some tinkering with the 'sudoers' file may be needed.

When the screen is sleeping for a while, it would be nice to turn the backlight off, and have it go back to the saved brighness when the screen is touched, and the system returns from sleep.

Some other 3rd party screen also support brightness control, but use different shell commands, so it would be nice to adjust the brightness control commands from the config.

Just putting the idea out there, I realise there are a lot of feature requests already. :)

UnchartedBull, 2020-02-10:
That would be something for custom actions. I try to make this work on as many displays as possible, so developing a slider which only works for a handful of devices might not be feasible, but for custom actions it does make sense. Putting this in the backlog.

You can turn off the display while sleeping with the `turnScreenOffSleep` attribute in the config.json. This will turn off the screen after I think 5 minutes being in sleep mode and will turn it back on, once you tap the screen. In the next release this can also be configured via the settings in the UI.

DeviousPenguin, 2020-02-10:
Thanks, I've seen other projects where the official RPi screen may be the easiest thing for backlight control initially, then other screens can be added in later on, if testers are avaliable.

I didn't know about the "turnScreenOffSleep" option, I've now set it to `true` in my config, and see how it works.

Currently my Octoprint never sleeps as I'm using the Klipper firmware. It doesn't matter is my printer is on or off, Octoprint and Klipper are always connected. I may be able to work around this however with some Delayed Gcode commands in order to make Octoprint disconnect from Klipper after 10 mins for example.

The only thing is I then need a way to make Octoprint reconnect to klipper, is it possible to get Octodash to send a message to Octoprint when Octodash wakes up, when the screen is touched?

UnchartedBull, 2020-02-10:
Great to hear, that this works for you. And yes the official Pi Screen is one of the easiest screens to work on and will be the first one supported by OctoDash. It may take some time though ...

DeviousPenguin, 2020-02-11:
No problem, even having the screen go off during sleep mode is good as I like to leave the Pi running even when not printing.

I have sometimes my screen sometimes goes off during printing, it does seem to go to sleep after a while, but a simple press of the screen makes the screen turn on again, and it goes back to showing the print progress.

I'm not sure if this is a bug/feature, but am happy to post logs if this isn't intended behaviour.

UnchartedBull, 2020-02-11:
I noticed similar behaviour once, although I don't use the screenOff feature, because I'm currently turning off the whole Pi with the 3D Printer. I think the problem here is, that I need to turn on dpms in order for the screen to turn off and I think turning dpms off doesn't work correctly. Could you paste the output off `xset q` here?

DeviousPenguin, 2020-02-13:
No problem, I have no keyboard on my Pi so I have SSH:

```sh
$ DISPLAY=:0 xset q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000000
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    off    02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  660    repeat rate:  25
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  2/1    threshold:  4
Screen Saver:
  prefer blanking:  no    allow exposures:  yes
  timeout:  0    cycle:  600
Colors:
  default colormap:  0x20    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/misc,built-ins
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  Monitor is Off
```

The screen is currently off as Octoprint is disconnected from Klipper, and my printer is off.

However I should say that after a few reboots and config changes, I am no longer having the screen turn of during prints, even though I have left `turnScreenOffSleep` set to `true` the entire time, so this is a tough bug to reproduce.

If it does start happening again I'll run the `xset` command again and record the output.

UnchartedBull, 2020-02-13:
The last 4 lines (DPMS) is the one that are most interesting. I think it doesn't get disabled correctly in some cases, but I still don't know why.

DeviousPenguin, 2020-02-14:
I'm not really sure either, the output of the `xset` command does seem to vary, it seems that DPMS may not be enabled just after a reboot, but is enabled after a few minutes of uptime.

Earlier today I rebooted the Pi, and the uptime was around 1 minute, and the last few lines of `xset` had this:

```
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Disabled
```

After an uptime of around 20 minutes, I have this:

```
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  Monitor is in Standby
```

I was only logged in using SSH the entire time, so maybe a service had a delayed start?

UnchartedBull, 2020-02-16:
DPMS shouldn't be enabled if you do nothing with the Pi. The Installation script turns DPMS off by default and OctoDash only turns it back on if it enters the sleep mode. 

Was OctoDash in Sleep Mode between those two outputs?

DeviousPenguin, 2020-02-18:
On a fresh boot Octodash does not go to sleep on my setup, unless I disconnect the link between Octoprint and Klipper, and Octodash usually stays in the main screen forever.

However occasionally on boot Octoprint fails to connect to Klipper as Klipper can take a while to startup, and in this case Octodash does go into sleep mode after around 5 minutes, as Octoprint is disconnected from the Printer (or klipper in this case). 

This is probably what happened here, now I know that it is OctoDash turning DPMS on then this makes for more sense. Sorry I can't explain this better, Klipper works in a very different way to other firmwares!

UnchartedBull, 2020-02-18:
I totally understand you. The main problem here is, that OctoDash should turn DPMS off, once it exists sleep mode, which does not seem to work reliable. I still don't know why, but I can reproduce this on my side ...

DeviousPenguin, 2020-02-19:
Ah, I see. I've not had it happen for a while, but next time it happens I'll try and grab some logs before I reboot or shut down the Pi. :)

UnchartedBull, 2020-10-21:
Request from @tomaae:

> Since there are many types of displays which are handled differently, I propose following solution:
> Add 2 more options to config:
> 
> Execute command in cli when screen is turned off
> Execute command in cli when screen is turned on
> For example in my case it would be:
> 
> `gpio -g pwm 19 0`
> `gpio -g pwm 19 40`

vvv850, 2020-11-26:
I'm using the 7" Rpi display and after Octodash "blanks" the screen the backlight is still on.

Is it possible to manually add the following to the sleep and wake up routine?
```
sudo sh -c 'echo "1" > /sys/class/backlight/rpi_backlight/bl_power'
sudo sh -c 'echo "0" > /sys/class/backlight/rpi_backlight/bl_power'
```

UnchartedBull, 2020-11-30:
currently not. Will be possible once this has been implemented.

UnchartedBull, 2021-01-12:
Maybe a small update: You can now customise the command that is being used to put the screen to sleep and to wake it up again. More info here: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#my-screen-doesnt-turn-off--on-while-octodash-is-sleeping

ammas09, 2021-01-17:
Any idea when this feature my be implemented?

I have a 7" Rpi display and am looking for a way to turn off the screen and backlight even if it's connected to the printer.  The sleep/auto shutoff works fine when it's sleeping but I want the screen to turn off even if I have a printer connected because my printer is on most of the time.

The 7" Rpi display only is rated for 20K hours so a little over 2years if left on continuously.  I have tried configuring screen blanking, screensavers, dpms outside of OctoDash but none of that seems to work.  It's like OctoDash is overriding those settings... 

Thanks for all of the work!  OctoDash is amazing I just don't want it to burn out my display needlessly.

vvv850, 2021-01-18:
I managed to get it working by updating the RPI eeprom and software.

UnchartedBull, 2021-01-23:
definitely the correct issue. With the next release you will be able to at least configure `xset` the way you like. Currently OctoDash will disable xset after waking up. If you change your wakeup command to just an empty string OctoDash will not overwrite the xset stuff on wakeup. Then you can just configure a normal screen blanking say every 15 minutes (this would happen during print as well though). I don't know when I'll have the time to tackle this issue, sorry. But probably something before / at v3.

mapotter99, 2021-06-14:
What is the status of all the topics linked to this issue?

Specifically, I have DPMS turned on with my Hyperpixel. The screen properly goes to sleep after about 5 minutes, and properly wakes with a touch, but the touch also registers as a button press in the OctoDash interface at the same time, which results in opening functions I didn’t intend to open. I’ve been sidestepping this by touching either the bottom left or bottom right corner, as there are no functions there on the OctoDash Home Screen.

s3frank, 2021-09-12:
I am running latest version of OctoPi and OctoDash with an RPi4 + Waveshare 7" DSI screen. It's hooked up to Prusa MK3Si+ and well the screen just never ever turns off unless I disconnect OctoPrint by pulling the USB cable out from the printer. Not the behavior I was hoping for. Can someone share a config and steps that allow you to make the screen go to sleep at all times after X seconds of no touching of the screen?

jneilliii, 2021-09-12:
OctoDash will only put the screen to sleep if your printer goes into a disconnected state. If your pi powers your printer and always stay connected then it will never go to sleep. This is known as backpowering, and can be resolved by either [taping the 5v pin](https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574?u=jneilliii), cutting the 5 volt cable in the USB cable, [power blough-r](https://www.th3dstudio.com/product/power-blough-r-pi-usb-power-blocker/) device or [similar](https://www.amazon.com/PortaPow-Cased-Power-Blocker-Single/dp/B094FYL9QT).

s3frank, 2021-09-13:
@jneilliii thanks for your reply. I guess what I am asking is either not possible with OctoDash then? I find that a bit weird to be honest. I think having a pattern similar to how other touch devices work makes sense or at least as a configurable feature. I respect that there are use cases where the Pi is managed as a separate system form the printer it is connected to, perhaps even more then 1 printer. In my use case. the Pi is integrated into the console of my Prusa using the Printerbox design,...basically enhancing my Prusa with a proper touch screen, permanently connected. In this use case screen power management like an iPad etc seems more appropriate?

Is there now way to do this? For example by turning off all screen power management by OctoDash and use something else outside of it to turn screen on and off based on a simple timer and last touch event?

Happy to experiment with others who want a similar thing.

If OctoDash stays on forever, it will wear my screen too fast and be a reason for switching to another UI that does support this if any exist. I really like OctoDash so again, happy to try and collaborate though I am not a Python coder (Java, Javascript and Shell scripts are ok).

jneilliii, 2021-09-13:
if the sleep commands work for you I added sleep custom actions to the companion plugin that allows you to press a button and put the screen to sleep, but those are currently only available while not printing.

UnchartedBull, 2021-09-15:
@s3frank it's already possible what you're trying to achieve here. See this issue: https://github.com/UnchartedBull/OctoDash/issues/710.

UnchartedBull, 2021-09-15:
Even during printing. This will be a fairly simple: turn off after x minutes if no touches have been registered.

If you only want to blank the screen if the printer is idle you either need to use the custom actions @jneilliii mentioned or wait until this issue is being fixed :)

bolsoncerrado, 2022-02-14:
> 

Hey @mapotter99 @UnchartedBull I'm also an Hyperpixel4 user and my displays stay ON  forever when printing. Did you sort this out so they turn off and only on after a touch of the screen?

Many thanks!

s3frank, 2022-02-14:
I followed this and it worked fine.
https://github.com/UnchartedBull/OctoDash/issues/710