UnchartedBull/OctoDash #430: Display stays black after boot

bmangodt, 2020-02-16:
**Describe the bug**
I have a RPi 3B+ with a Pimoroni Hyperpixel 4.0 Touchscreen display; which is a display that uses the headers on the pi. I installed both the drivers for the touchscreen (which is working fine to my knowledge) and OctoDash. The Pi boots up, logs in automatically, ratpoison shows a small splash screen and then... nothing but black... I had OctoDash running 1 time in the multiple times I rebooted trying to solve the issue. 

**To Reproduce**
Steps to reproduce the behavior:
1. Boot Pi 3B+ with Hyperpixel attached
2. Display shows boot logs and logs in to default user
3. Ratpoison shows up in top right corner
4. Display stays blank

**Expected behavior**
OctoDash shows on the display

**General Information:**
 - Raspberry Pi 3B+ and [Pimoroni Hyperpixel 4.0 ](https://shop.pimoroni.com/products/hyperpixel-4?variant=12569485443155)
 - Octopi Version v0.15.1
 - OctoDash Version latest from Git
 - OctoPrint Version v1.3.12

UnchartedBull, 2020-02-16:
Could you try logging in via SSH and setting your display (`set DISPLAY=:0`), after that execute `octodash`. Please report back if that works, if not, please post the errors here.

bmangodt, 2020-02-16:
Tried that already... octodash starts up normally (I think) but still the display remains black. Here's the output: 

`Fontconfig warning: "/etc/fonts/fonts.conf", line 100: unknown element "blank"
Fontconfig warning: ignoring UTF-8: not a valid region tag
ATTENTION: default value of option force_s3tc_enable overridden by environment.`

**NOTE:** I tried a fresh install of octopi (0.17) with no plugins, just basic configuration and it works fine. Could it be that I already had one or more plugins installed that Octodash uses? I had all of them excluding the DisplayLayerProgress. I used a similar one but disabled it in Octoprint settings and installed the one needed using the dependancies from the install script.

UnchartedBull, 2020-02-17:
Are you sure that your display is running on :0? Could you maybe check how many displays are connected to the Raspberry? Maybe you have HDMI Hotplug enabled which may create another display? 

The warning shouldn't be something to be concerned about, but if there is no error OctoDash should start up normally ...

bmangodt, 2020-02-17:
That was probably the issue here, yes. I stopped looking for a solution and merged over my settings to the new install (way easier haha). Thanks for the help!

closed 2020-02-17 by bmangodt