UnchartedBull/OctoDash #433: Turning the printer off and reconnecting via OctoPrint leaves OctoDash in a weird state

UnchartedBull, 2020-02-18:
All OctoDash Versions

steviehs, 2020-02-18:
Hi, good news, you added that :-)
My printer (CR-10) is switched on by a WIFI Plug from a OctoPrint Plugin (Tasmota). After that OctoPrint connects automatically to the printer. So my perferred behaviour would be, that OctoDash shows an idle screen (or even better a On Button for the printer :-) while not connected and then goes to the standard dash when connected without any alerts, as loosing and getting a connection is (normally) intended.
If I can help with any testing, let me know.

UnchartedBull, 2020-02-24:
That is already supported with the latest version ... Once the printer is disconnected OctoDash will enter a Standby Mode, tapping the screen will tell OctoPrint to reconnect to the printer. With 1.4.0 you could also use the PSUPlugin to turn on your outlet (via the settings menu). This will tell the PSUPlugin to turn on the outlet, wait 5 seconds and then try to connect to the printer. The Tasmota plugin isn't supported though, but I think you can switch that around.

steviehs, 2020-03-13:
Great, had now time to update and switch back to PSU-Plugin. Everything works great. Now wishing for automatic reconnect to printer and - if PSU-Plugin and switch on from sleep screen is active - a nice icon on the sleep screen but ey :-) Thanks a lot for your work!

UnchartedBull, 2020-03-13:
Great that this is working for you! Automatic reconnect is something for v2.x :)

PR #532 (2020-03-31): fix/reconnect-after-print-failed
Fixes #433.
closed 2020-03-31 by UnchartedBull