UnchartedBull/OctoDash #445: Update message popping up even though I am on the latest version

spiff72, 2020-02-21:
**Describe the bug**
The update notification message is popping up when I am using the newly released 1.4.0 version.

**Expected behavior**
Since I am current, I wouldn't expect to see this notification.

**Screenshots**
[image]

**Additional context**
This may be a special case, as I updated over top of the modified version when you were working on a fix for the saving of settings black screen of death.

spiff72, 2020-02-21:
I updated by downloading the deb file from the assets section of the Release 1.4 tab, and used the dpkg command to perform the update. 

I did this because I wasn't sure if the update script would work correctly with my "modified" version of the package.

evannadeau, 2020-02-21:
I see this same behavior, however I also had a little bit of a strange upgrade. I was following the upgrade instructions from the wiki and blindly posted in end download line that pulled down version 1.0.0. so I went from 1.3.3 to 1.0.0 and then up to 1.4.0

DeviousPenguin, 2020-02-21:
Getting the same thing also:

[pic](https://i.imgur.com/uP9yuFFl.jpg)

UnchartedBull, 2020-02-22:
I think I messed up something with the version check there ... Will update once I get back home. Sorry for that!

UnchartedBull, 2020-02-24:
Well, I did something really stupid this time around. Accidentally labeled the GitHub release v.1.4.0, see that point in front of the 1. That is what threw everything off. Good thing is, that this is now fixed, just needed to update the GitHub Release name, no update needed for OctoDash. Thanks for pointing this out!

closed 2020-02-24 by UnchartedBull